ricaun/DuinoCoinI2C #2: Free Ram

modfiles, 2021-05-20:
Running the rig for sometime will result in Free Ram only appearing in console and it will stop mining

ricaun, 2021-05-20:
I had a similar some problem and I update some code last night to try to fix.

But I still have some strange while loop problem...

How many slaves are y using?

modfiles, 2021-05-20:
im just using three nano for a test. did reprogram esp8266 and nano, re-wire, change power supply to highee current one but to no avail

mmetzger1234, 2021-05-22:
Same issue. will run for awhile then I will check on it and  nothing is mining.  have 4 arduino nanos tied to a ai-tinker esp8266 nodemcu. i was going to play around with code to see if anything jumps out that would cause this. .

ricaun, 2021-05-22:
Yes something is not working....

I update the code and add some features to make it more reliable, 

Now the Serial monitor shows the is connected on the I2C, like below...

```
[ ]FreeRam: 45976 I2C Connected[ 1 2 3 4 5 ]
```

And I'm forcing to stop the connection when a BAD is receive. 

I had some problems with some boards to always sends a BAD response and the server block the connection for some time.

closed 2021-08-07 by ricaun