ricaun/DuinoCoinI2C #3: About the Wiring (Question)

PipeDeveloper, 2021-05-21:
Hello!

Hello! I would like to try this project by connecting several arduino mini boards in parallel, but i'm wondering about the best way to power each board

In schematics says that arduino works with 5v and ESP8266 works with 3.3v
I see that you power arduino on the 5v pin (directly to the chip bypassing all regulators) but with 3.3v That actually works fine? (low voltage means overclock the internal processor).

Another question is if you power ESP8266 via USB, then you are using the wemos D1 mini regulator 5v to 3.3 to power the ESP8266 and all ATMegas chips, so... you are stressing that regulator to its max capacity (with more devices you plug on 3.3v output).

On this case, i'm wondering which is the best way to make the connection optimizer.
Im thinking to use a phone charger 5v 2A to power directly all on 5v (includding the Wemos D1 Mini). As i2c signals on arduino are 5v, and ESP8266 signals are 3.3 (prefered, 5v is also supported). Use a voltage divider resistor 2K and 1K to drop 5v to 3.3 (here i don't know exactly if i sould do on SCL, SDA or both) or maybe use a logic level bidirectional.

The other option is to use a lm2596 module, connect the input to a phone charger 5v/2A or universal charger 12v/2A and regulate the output to 3.3v and power all directly just as the schematic (but on this way it doesnt like as arduino works with 5v).

Please comment wich is the best way to power up many boards. Obviusly we always think in the most simple way and if it works it doesnt matter, but my goal is to learn, understand and do the "correct" way to dont damage any component at long term.

ricaun, 2021-05-21:
Hello @PipeDeveloper 

> In schematics says that arduino works with 5v and ESP8266 works with 3.3v
> I see that you power arduino on the 5v pin (directly to the chip bypassing all regulators) but with 3.3v That actually works fine? (low voltage means overclock the internal processor).

Yes, the Arduino can handle 3.3V and 16MHz, the datasheet does not approve this but works. This a cool trick! [Arduino 3.3v-conversion](https://learn.adafruit.com/arduino-tips-tricks-and-techniques/3-3v-conversion).

> Another question is if you power ESP8266 via USB, then you are using the wemos D1 mini regulator 5v to 3.3 to power the ESP8266 and all ATMegas chips, so... you are stressing that regulator to its max capacity (with more devices you plug on 3.3v output).

Yes probably is stressing the regulator and should be limited the number of Arduino connected to this power supply.

> On this case, i'm wondering which is the best way to make the connection optimizer.
> Im thinking to use a phone charger 5v 2A to power directly all on 5v (includding the Wemos D1 Mini). As i2c [...]

Power everything with USB should be a better design, and a logic level bidirectional is the best solution, but I gonna try with some resistor and make a voltage divider with resistor 2K and 1K, this part needs to be improved.

> The other option is to use a lm2596 module, connect the input to a phone charger 5v/2A or universal charger 12v/2A and regulate the output to 3.3v and power all directly just as the schematic (but on this way it doesnt like as arduino works with 5v).

Power everything with 3.3V should be fine.

> Please comment wich is the best way to power up many boards. Obviusly we always think in the most simple way and if it works it doesnt matter, but my goal is to learn, understand and do the "correct" way to dont damage any component at long term.

In my opinion, the easiest way is powering everything with 3.3V, but if you have a lot of boards you should use a regulator lm2596 to supply all the power with 3.3V.

Another way is to power the Arduino with 5V like it's should be the normal way to work, and add a logic level bidirectional or resistors to connect between the Esp and the Arduinos on the I2C bus.

I pretend to add the second option on the documentation.

PipeDeveloper, 2021-05-21:
Thank you very mcuh @ricaun 

For bidirectional level shifting i found this video very interesting
https://www.youtube.com/watch?v=t-yuYasIKtY&ab_channel=KevinDarrah
(See the comparison signal at the end of the video)

I also think that power all with 5v should be the best option, but avoiding power it with USB (since the USB input goes through a linear DC-DC converter 5v at max 1A, and there is loss of energy here, like 85-90% efficiency), in stead power directly on the 5v pin of the arduino (the same ATMega chip have an internal regulator that tolerates a range of 4,85 and 5,20v) thats the operational range, so if you power it with a phone charger (not using the USB) should be the best way, more eficient and also you have available the max output current that the phone charger could handle 2A 2.4A most of them or even 3A if you use a raspberry pi transformer

And i guess using a mosfet like [this](https://www.sparkfun.com/products/12009) should be the best option (all depends on the frequency of the signal) but using a voltage divider resistors should be the worse option, all depending on how many devices you connect.

closed 2021-09-14 by PipeDeveloper