ricaun/DuinoCoinI2C #6: Feature Request - Ideas

PipeDeveloper, 2021-06-04:
Hello

I would like to suggest some features to improve and makes more easier to setup this proyect for people that loves do this kind of stuffs or people that just want to setup the easier and simplier ways

**1.- Web Server (port 80)**: This feature is for monitoring the rig with some flexibility like to allow change the username of DUCO mining, a buttom to start/re.start/stop (maybe add an automated reboot feature every "X" time, hour) to improve the  mining performance. Also allow to visualize all devices on the I2C with their respective adress and status (if they are mining correctly).

**2.- Wifi AP mode**. This is for the first boot, in stead to define the SSID and password of the wifi network on the main code, have the flexibility to imput at the first boot, you you connect with your phone, enter (or search) your network and then the device will connect to your network. Also define a buttom to reset the values, so if you change SSID, you dont have to load the code again.

**3.- A compiled firmware for ESP8266**. This is to complie all code and libraries on a .bin file ready to upload to your ESP8266 using a PyFlasher. This will be more easier than using arduino IDE. Also for future updates, you may update wireless your ESP8266 by uploading .bin file. Also you can do a "release" of this compilance.

Would be nice to have those features. Also will help to improve the performace, detect and fix some bugs in the way.
I see the potential of this proyect and im glad to contribute but i dont have programming skills to develop all.

ricaun, 2021-06-04:
Thanks for the ideas:

I'm working on something like this, at the moment is a console.

[image]

I'm still trying to figure out what's going on with some slaves, who always return a bad job.

Firmware for ESP8266 is great, when I add the Wifi AP mode I gonna create firmware to make more simple to upload on the esp.

PipeDeveloper, 2021-06-04:
Yeah! a console also is nice!! maybe for an advance config (or to input commands) or see a "level message" such debug messages and share submit (as you did) is a very nice work.

There is a proyect called "Tasmota" used commonly for domotic (such control relays for home lights, sensors and many other things). But you may use their libraries since they have all the features that i mentioned (includding the console) and many other stuffs.

https://github.com/arendst/Tasmota

The program is very stable (i use for my home automatization actualy) and you could use some of its code since all it is under general public licence.

Dark-Manu, 2021-06-16:
Nice ideas!
I was thinking about using an HC-12 to forward the serial exit to another arduino/device and read it from here. As I use this on my mini-weather-station, I though that it could be possible.

[image]

Anyway, any kind of monitoring system would be great for people like me that plan to run my rig solar-powered.