ricaun/DuinoCoinI2C #7: Server Behaviour

Dark-Manu, 2021-06-05:
Hello,
Let me say first that it's a really great project because it permits to use a master and several slaves without a PC and therefore it's possible to use low powering devices.
However, I'm wondering if could be possible to make the behaviour of the server, (ESP8260/ESP32) to act as a PC client AVR miner. I say this because the miner (for the server) appears like it is, (an ESP8266 I2C v2.4.7) but working on a low level difficulty due that the job is being done by an arduino.
I'm not 100% sure about this, because I'm just starting with Duco but, do you thin that the server could penalyze the miners working this way?

Not related but, only to let you know that arduino pro mini, that runs on 3.3 volts, works great as slave with a Wemos D1 mini as a master, and this can avoid voltage issues when using 5 volts board.

Great Job!

PipeDeveloper, 2021-06-06:
Good observation.
This is important, because if the server detects as one device (with low level difficulty) no matter the device it is, if it's not an arduino (AVR chip) server will penalyze the reward.

An ESP8288 does the power of 20-40 arduinos (depending the frecuency). So no matter how many arduinos you attach to I2C, if the server recognizes as just the ESP8266, the reward will be very poor.

ricaun, 2021-06-07:
When the esp request a job I'm sending like an Arduino (AVR)
https://github.com/ricaun/DuinoCoinI2C/blob/f6ba5d6053388f3495ae7320fe397b735b1dfe27/DuinoCoin_Esp_Master/DuinoCoin_Esp_Master.ino#L32

The server responded with a difficulty based on this request, and if this job is finished too fast the server knows is not an AVR board and sends a bigger difficulty.

I believe the name of the MINER does not have any influence on the performance.

At the moment I'm having 7 IC2's and have ~35.53 ᕲ daily profit.

Dark-Manu, 2021-06-08:
Yes, you are right. Checking mi stats with more detail I see that, although the device is detected as ESP8266, all the received jobs have difficulty level "6". The only problem I have to face now is that my device have a lot of disconnections. At this time I have 3 nanos connected to a Wemos D1 mini and following your wire schema... I need to do more test
thanks

PipeDeveloper, 2021-06-08:
maybe you need to power your arduinos with 5v, and use a level shifter converter on the I2C, HV for arduino and LV to your ESP8266. I have not test it yet, but maybe is something on the signal or stability.

ricaun, 2021-06-08:
I'm testing this option with a level shift converter! Works great!
The main esp code needs to be improved, I'm testing a different version.

Dark-Manu, 2021-06-08:
> 
> 
> maybe you need to power your arduinos with 5v, and use a level shifter converter on the I2C, HV for arduino and LV to your ESP8266. I have not test it yet, but maybe is something on the signal or stability.

I tested it with a Wemos D1 board (not a mini) that can be powered with 5 volts so, I could power the arduinos with 5 volts but also I had the same problem with disconnections. I tend to believe that the server sometimes are too busy to process all the requests... not sure about that.

But the truth is that I like this project a lot and I'm willing to do all the needed tests until I get all my boards working as expected.

PipeDeveloper, 2021-06-08:
Wemos D1 R2 has the same ESP8266, the only difference is the board size, but anyways should work the same as wemos D1 mini. But before start coding, you should discard any hardware related issues, for example avoiding using this kind of connector or jumper wires:
[image]

Since the low quality and high resistance can modify the signals. Most common issues about long distances or many devices on the I2C bus are that you need to add 10k ohm resistor to pull up signals over the circuit (some devices had it on their build like the ESP8266 the entire chip, but not wemos d1 mini pro and some lora boards are the exception) and raspberry pi but not the arduino boards. If you use a logic level converter, it has incorporated 10k pull up resistors so you are good to go for aprox 20 devices with no issues.

Next check your wifi and internet stability. As i know the server has a timeout for re-connection, so maybe add a delay in the code for wait server response. 5-10seconds should be enough for the slowest connection or high ping. I can imagine people like me on southamerica have 250-300 ms of ping and maybe it can have an impact on the stability.

Dark-Manu, 2021-06-15:
> Wemos D1 R2 has the same ESP8266, the only difference is the board size, but anyways should work the same as wemos D1 mini. But before start coding, you should discard any hardware related issues, [...]

Definitely, you are right. I was using these cheap dupont connection wires on my setup so I replaced these with solid copper wires and the performance improved a lot. There is mi test-rig
[image]

This time I made some test using a Wemos D1 mini and 3 Arduino Nano and for the results, al worked well when running only 1 arduino + Wemos or 2 arduino + Wemos, but when I run the three arduinos at the same time, ig goes well for the first 1 to 3 hours, but later I lost one of the nanos, (getting BAD jobs or disconnecting).

Next step: I've added two pull-up resistor, (10K) and and got the system up and running. Now I only need to wait some hours to see if the resistors have any effect.

ricaun, 2021-06-16:
Hello! I updated the Esp Master Code to make it more reliable.

Dark-Manu, 2021-06-16:
> 
> 
> Hello! I updated the Esp Master Code to make it more reliable.

Nice!, I'm going to install it right now.
I'll let you know how was in my rig
Thanks :)

Dark-Manu, 2021-06-28:
Hi,
Only to let you know that I've just installed the last DuinoCoin_Esp_Master and, probably because of the server issues I'm having several "time outs". At least, I can mine!
Keep testing

19:53:21.339 -> [ ]FreeRam: 43888 I2C Connected[ 0 1 2 3 4 ]
19:53:22.342 -> [0]Job Done: (596) Hashrate: 220.54
19:53:26.300 -> [ ]FreeRam: 44560 I2C Connected[ 0 1 2 3 4 ]
19:53:26.801 -> [4]Job Receive: 6
19:53:28.475 -> [4]Job Done: (145) Hashrate: 216.24
19:53:31.314 -> [ ]FreeRam: 44560 I2C Connected[ 0 1 2 3 4 ]
19:53:36.327 -> [ ]FreeRam: 44560 I2C Connected[ 0 1 2 3 4 ]
19:53:41.315 -> [ ]FreeRam: 44560 I2C Connected[ 0 1 2 3 4 ]
19:53:46.330 -> [ ]FreeRam: 44560 I2C Connected[ 0 1 2 3 4 ]
19:53:48.436 -> [2] --------------- TIMEOUT ------------- 
19:53:50.287 -> [3] --------------- TIMEOUT ------------- 
19:53:51.343 -> [1] --------------- TIMEOUT ------------- 
19:53:51.343 -> [ ]FreeRam: 44480 I2C Connected[ 0 1 2 3 4 ]
19:53:52.346 -> [0] --------------- TIMEOUT ------------- 
19:53:56.305 -> [ ]FreeRam: 44344 I2C Connected[ 0 1 2 3 4 ]
19:53:58.510 -> [4] --------------- TIMEOUT ------------- 
19:54:01.319 -> [ ]FreeRam: 44208 I2C Connected[ 0 1 2 3 4 ]
19:54:07.489 -> [0]Connecting to Duino-Coin server...
19:54:12.704 -> [0]Connection failed.
19:54:12.704 -> [ ]FreeRam: 44152 I2C Connected[ 0 1 2 3 4 ]
19:54:17.719 -> [ ]FreeRam: 44264 I2C Connected[ 0 1 2 3 4 ]
19:54:22.679 -> [ ]FreeRam: 44264 I2C Connected[ 0 1 2 3 4 ]
19:54:27.694 -> [ ]FreeRam: 44264 I2C Connected[ 0 1 2 3 4 ]
19:54:32.709 -> [ ]FreeRam: 44264 I2C Connected[ 0 1 2 3 4 ]
19:54:37.523 -> [0]Connecting to Duino-Coin server...
19:54:37.575 -> [1]Connecting to Duino-Coin server...
19:54:37.675 -> [2]Connecting to Duino-Coin server...
19:54:37.776 -> [3]Connecting to Duino-Coin server...
19:54:37.823 -> [4]Connecting to Duino-Coin server...
19:54:37.976 -> [ ]FreeRam: 43464 I2C Connected[ 0 1 2 3 4 ]

Dark-Manu, 2021-06-28:
Little update:
You probably noticed that. Althoug the arduinos are sending valid jobs, these jobs are not counting on the wallet, the balance remains the same :(

ricaun, 2021-06-29:
Yes something change and it's not working 😞

ricaun, 2021-08-07:
The code was updated to duinocoin version 2.6.
Now everything is working fine. 
Download and reupload the firmware.

closed 2021-08-07 by ricaun