ricaun/DuinoCoinI2C #9: I cannot upload slave code into my arduino

MudaMuda2021, 2021-06-28:
Dear All, 
I am a rookie, and mining duino-coin sounds interesting to me. So, I downloaded the code (Master & Slave), and I uploaded master code into ESP8266 with successfully. But when I tried to upload slave code into my arduino Uno and error message showed as below: 
***************************************************************************************
DuinoCoin_Wire:8:10: fatal error: DuinoCoin.h: No such file or directory
 #include <DuinoCoin.h>        // https://github.com/ricaun/arduino-DuinoCoin
          ^~~~~~~~~~~~~
compilation terminated.
exit status 1
DuinoCoin.h: No such file or directory
***************************************************************************************

Anyone please knowledge me what I did wrong? Or did I miss any step? 

Sincerely, 

MudaMuda

ricaun, 2021-06-28:
You need to download the library: https://github.com/ricaun/arduino-DuinoCoin and install on your Arduino IDE.

This library has the encrypt worker used on the duino job.

closed 2021-06-28 by ricaun

MudaMuda2021, 2021-06-28:
ricaun, 
Thanks for sharing. Problem solved. I finally figure out. There are a few requested libraries that I did not add in my arduino IDE.
After adding those library, I can upload my slave code into my arduino. Cannot wait to play it. 

Thanks, 

MudaMuda2021