ricaun/DuinoCoinI2C #11: Random Specified Ports

modfiles, 2021-07-02:
Any possibility to implement the joggling ports like the new feature of the official miners? and maybe an option to ping the server ip+ports if it will response, otherwise will choose on list of specified ports and just a quick question, to what line does it fetch or display the server version?

ricaun, 2021-07-02:
I implemented exact that yesterday.
I added an array of available ports and each connection tries a port and goes to the next and so on.

modfiles, 2021-07-02:
thanks, i happened to see the update and tested it but it stuck in connecting to server and connection failed loop for hours but prior to testing the latest update, all ports are working so atm, i switch to previous version

emulatormania, 2021-07-03:
cant connect with random ports

ricaun, 2021-07-03:
Thanks @rassware to fix the random port connection.

closed 2021-07-03 by ricaun