ricaun/DuinoCoinI2C #13: Error compiling with other AVR Boards

PipeDeveloper, 2021-07-08:
I have some Atmega8 chips. but i got this error when compiling.

c:/users/felipe rojas verasay/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: C:\Users\FELIPE~1\AppData\Local\Temp\arduino_build_297722/DuinoCoin_Arduino_Slave.ino.elf section .text' will not fit in region text'
c:/users/felipe rojas verasay/appdata/local/arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: region `text' overflowed by 3684 bytes
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compilando para la tarjeta ATmega8.

PipeDeveloper, 2021-07-10:
On Atmega8
Flash size 7680 bytes.
Global 1024 bytes.

The sketch uses 11620 bytes of flash and 723 bytes on global.

There is any way to compress this code to save flash memory?

ricaun, 2021-07-20:
Good question, I do not design the code to be compact.

The idea was to use the Arduino Uno (atmega328p)... That's why I use some external libraries to make the code more clear and bigger.

PipeDeveloper, 2021-07-20:
It would be nice to support atmega8 for those alternative arduinos and diy projects.
You can make a branch to dont affect this project, or another repository, im glad to help you in what i could. Also I have the hardware to test it.

Maybe a tip could be remove the i2c library and asign a fixed i2c address (to save flash memory) but be concious that in the code you upload to every arduino you must change the i2c address to a different one. (maybe a commented lines wit a few suggested address).
Another tip could change the pin-out pin-in for #define port. That also save some flash memory.

I dont know if it is enough to make the code smaller, but it's a beggining

ricaun, 2021-07-20:
> 
> 
> It would be nice to support atmega8 for those alternative arduinos and diy projects.
> You can make a branch to dont affect this project, or another repository, im glad to help you in what i could. Also I have the hardware to test it.
> 
> Maybe a tip could be remove the i2c library and asign a fixed i2c address (to save flash memory) but be concious that in the code you upload to every arduino you must change the i2c address to a different one. (maybe a commented lines with a few suggested address).
> Another tip could change the pin-out pin-in for #define port. That also save some flash memory.
> 
> I dont know if it is enough to make the code smaller, but it's a beggining. As many people said, if the code is simple, is better.

I don't know if the main https://github.com/revoxhere/duino-coin has an atmega8 portability. Probably is easy to add the i2c code on the small code. 

7680 bytes, is hard 😢

PipeDeveloper, 2021-07-21:
You can take a look to
https://github.com/kickshawprogrammer/ATTiny85-DUCO-Miner

He made an small code for duco-coin library, the whole code takes about 5000 bytes. The thing is that Attiny85 uses just one wire signal, while i2c is pretty much faster.

ricaun, 2021-08-07:
I'm trying to make the code small, the program is the available SRAM. I manage to set the code to around 5000 bytes the ram is almost 500.

I gonna test on a tiny85 and check if works fine.

PipeDeveloper, 2021-09-14:
Thank you @ricaun i'll close this thread. Project running like a charm 👌

closed 2021-09-14 by PipeDeveloper