ricaun/DuinoCoinI2C #15: Can Raspberry Pi Pico be used as slave?

airtikit, 2021-07-11:


sunsunny83, 2021-07-13:
I've tried this but not working.....

ricaun, 2021-07-20:
Should be possible, but is needed to update the library - https://github.com/ricaun/arduino-DuinoCoin

But I don't know if the I2C slave code gonna work.

airtikit, 2021-09-12:
Error compiling for board Raspberry Pi Pico, "ArduinoUniqueID".

[image]

airtikit, 2021-09-12:
> Should be possible, but is needed to update the library - https://github.com/ricaun/arduino-DuinoCoin
> 
> But I don't know if the I2C slave code gonna work.

Error compiling for board Raspberry Pi Pico, "ArduinoUniqueID".

[image]