ricaun/DuinoCoinI2C #16: Update code to the latest AVR code for better stability?

Josstri, 2021-07-16:
I was wondering where the core code for the AVR code is stored? And how easy is it to update the code to the newest version? My atmega328p keep getting kicked off due to too many bad shares.

ricaun, 2021-07-20:
The code of the handle the `Ducos1a` is on the library - https://github.com/ricaun/arduino-DuinoCoin

I believe the problem is not on the library, my i2c implementation has some problems... 

Somethimes a slave goes to some BAD loop and stop work, just return BAD jobs...

Josstri, 2021-07-20:
Good to know. I will keep experimenting to see if it is a configuration issue

Josstri, 2021-07-20:
Might as well ask, what is the known maximum Arduinos that could be run at 5v with a bi directional data converter? I know the theoretical is 10.

ricaun, 2021-07-20:
> 
> 
> Might as well ask, what is the known maximum Arduinos that could be run at 5v with a bi directional data converter? I know the theoretical is 10.

Good question, the default is 10. Is possible to set like 100 but I don't know if the Esp has the memory for that and I don't know if the Esp has the speed to handle all the slaves with a good speed, probably not.

Josstri, 2021-07-20:
I was thinking that maximum I would only want to attempt 12 on one esp. I might have to experiment. I would just change it to #define clients 12, correct?

ricaun, 2021-07-20:
> 
> 
> I was thinking that maximum I would only want to attempt 12 on one esp. I might have to experiment. I would just change it to #define clients 12, correct?

Yes only change the `#define` the code makes the rest, I create the code scalable. 😄

Josstri, 2021-07-22:
Interesting issues now, when I have more than 6 devices, it is hard for them all to start working together. They don't blink when they are turned on. They are being seen by the esp, and being recognized, but they don't seem to be hashing

Josstri, 2021-07-22:
[image]