ricaun/DuinoCoinI2C #19: ive done everything in my knowledge to fix it

spareartsa, 2021-07-20:
[image]

ricaun, 2021-07-20:
Probably some connection on the I2C is not right, you should test with just one and see that happen.

spareartsa, 2021-07-20:
[image]

ricaun, 2021-07-20:
Something is bad with the I2C, the `Job Done:` should not return `(0)`.

Try to recompile the slave.

spareartsa, 2021-07-20:
[image]

spareartsa, 2021-07-20:
join general in the server

ricaun, 2021-07-20:
Something is bad on the I2C connection... The Connected should return always [ 0 ].

spareartsa, 2021-07-27:
do you think putting all the arduinos on one crystal oscillator will fix it. cuz thay go out of sync

ricaun, 2021-08-07:
> do you think putting all the arduinos on one crystal oscillator will fix it. cuz thay go out of sync

What? Who says that?

Check you i2c connection, you should re update the code and check wires and pins connection.

closed 2021-08-07 by ricaun