ricaun/DuinoCoinI2C #20: ATMega328 change to difficulty @16

PipeDeveloper, 2021-07-22:
ATMega328 with the optimized duino-coin library do about 215-220 H/s, however the server interpret the device as the arduino mega.

The server interpret 190 H/s arround as an standard arduino with dificulty of @6
But with this code 220H/s server rise to dificulty of @16, affecting the reward (which it pays less) and the performance, less submits.

In the few days the performance of the reward has been drop in about 30-40% when usually i do 80-90 DUCO per day, now i do about 40-50 DUCO per day with the same setup.

ericddm, 2021-07-22:
I have the same problem as you, and what if i add 100ms delay to ATMega328？ 
I am gona try it

ericddm, 2021-07-22:
[image]

ktannady, 2021-07-22:
does it work?

ericddm, 2021-07-22:
> does it work?

https://github.com/revoxhere/duino-coin/issues/681
revoxhere has fixed that bug :)

ktannady, 2021-07-22:
wow thanks! BTW really nice board!

ricaun, 2021-07-22:
@ericddm nice board! Really cool!

@PipeDeveloper About the 215-220 H/s speed... The library uses a diferente encoder then the original project, I gonna create a new version on the library using the original encoder.

The job difficult probably gonna stay on 6.

ricaun, 2021-08-07:
I update the library to the default encrypted, and the server now uses difficulty 4.

closed 2021-08-07 by ricaun

Dendrobats, 2021-09-03:
This problem is back again on 2.7

ricaun, 2021-09-03:
Yes something change on the server, gonna make some tests...