ricaun/DuinoCoinI2C #21: Trouble with Arduino slave

spooniestfiend420, 2021-07-22:
I saw a similar resolved issue to this but I can't figure this one out. When uploading the docs to make my Arduino nano a slave I get this error. "Fatal error Arduino.h no such file or directory .... Include arduino.h" 
I'm assuming that's a library I need to add to the ide but I can't find it. I found a file arduino.h in the Arduino download but it's where it was supposed to be in files and I can't upload it as a library on ide so I'm stumped. 
Any help appreciated.

PipeDeveloper, 2021-07-22:
arduino.h is related to the arduino board. Before upload the code you need to select the correct board. Select arduno nano and then select the based microcontroller ATMega328 for the original or ATMega328 (old bootloader) for the copy.

spooniestfiend420, 2021-07-22:
> arduino.h is related to the arduino board. Before upload the code you need to select the correct board. Select arduno nano and then select the based microcontroller ATMega328 for the original or ATMega328 (old bootloader) for the copy.

Yeah I did select the nano and I started with the old bootloader per the instructions that came with my Arduino and when that didn't work I tried the new version just to be sure