ricaun/DuinoCoinI2C #22: Mega328 have same unique ID

ericddm, 2021-07-24:
I have 1 ESP8266 and 6 Arduino Nano. But I only get 11-12 Duino Coins per day,  and I donot know why.

I find the 6 Arduino Nano print same unique ID after their job has been done.

I donnot know if it is working good.

ricaun, 2021-07-31:
The Atmega328p does not have a unique id on the datasheet, some counterfeit MCU has the same id.

https://github.com/ricaun/ArduinoUniqueID/blob/master/MCU.md#Disclaimer

I don't know if the DuinoCoin server gives less Duino Coins if you have the same id.

closed 2021-07-31 by ricaun

ericddm, 2021-08-01:
> The Atmega328p does not have a unique id on the datasheet, some counterfeit MCU has the same id.
> 
> https://github.com/ricaun/ArduinoUniqueID/blob/master/MCU.md#Disclaimer
> 
> I don't know if the DuinoCoin server gives less Duino Coins if you have the same id.

I have 6 Arduino ATMEGA328P, and I can get 10 DuinoCoins a day,  is that normal speed?