ricaun/DuinoCoinI2C #27: TIMEOUT How can I  fix this

fellboy42, 2021-07-29:
Hi 
whas working fine 2 days ago I have up loaded again and still get this

[0]Connecting to Duino-Coin server... with port 2814
[1]Connecting to Duino-Coin server... with port 2816
[ ]FreeRam: 47184 I2C Connected[ 0 1 ]
[ ]FreeRam: 47184 I2C Connected[ 0 1 ]
[ ]FreeRam: 47184 I2C Connected[ 0 1 ]
[ ]FreeRam: 47184 I2C Connected[ 0 1 ]
[ ]FreeRam: 47184 I2C Connected[ 0 1 ]
[0]2.5
[0]Job Request: ********
[ ]FreeRam: 46512 I2C Connected[ 0 1 ]
[1] --------------- TIMEOUT -------------
[ ]FreeRam: 47592 I2C Connected[ 0 1 ]
[ ]FreeRam: 47592 I2C Connected[ 0 1 ]
[ ]FreeRam: 47592 I2C Connected[ 0 1 ]
[ ]FreeRam: 47592 I2C Connected[ 0 1 ]
[ ]FreeRam: 47592 I2C Connected[ 0 1 ]
[ ]FreeRam: 47592 I2C Connected[ 0 1 ]
[0]Connecting to Duino-Coin server... with port 2813
[1]Connecting to Duino-Coin server... with port 2814
[ ]FreeRam: 47184 I2C Connected[ 0 1 ]
[ ]FreeRam: 47184 I2C Connected[ 0 1 ]
[ ]FreeRam: 47184 I2C Connected[ 0 1 ]
[ ]FreeRam: 47184 I2C Connected[ 0 1 ]
[ ]FreeRam: 47184 I2C Connected[ 0 1 ]
[1]2.5
[1]Job Request: ********
[ ]FreeRam: 47376 I2C Connected[ 0 1 ]
[1]Connecting to Duino-Coin server... with port 2814
[0] --------------- TIMEOUT -------------
[ ]FreeRam: 47376 I2C Connected[ 0 1 ]
[ ]FreeRam: 47376 I2C Connected[ 0 1 ]
[ ]FreeRam: 47376 I2C Connected[ 0 1 ]
[ ]FreeRam: 47376 I2C Connected[ 0 1 ]
[ ]FreeRam: 47376 I2C Connected[ 0 1 ]
[1]2.5
[1]Job Request: *******
[ ]FreeRam: 47592 I2C Connected[ 0 1 ]
[0]Connecting to Duino-Coin server... with port 2816
[1]Connecting to Duino-Coin server... with port 2813
[ ]FreeRam: 47184 I2C Connected[ 0 1 ]
[ ]FreeRam: 47184 I2C Connected[ 0 1 ]

panayotovip, 2021-07-29:
From several days i do have that problem too.

ricaun, 2021-08-07:
The code was updated to DuinoCoin version 2.6, this problem was fixed, please download and reupload the firmware.

closed 2021-08-07 by ricaun