ricaun/DuinoCoinI2C #29: need update to revoxhere/duino-coin v2.6

panayotovip, 2021-07-30:
After update from revoxhere/duino-coin to v2.6 stand alone nanos work no problems.
Here is running AVR I2C v2.5.2 and it is a lot of troubles with nodes.

Please do update and add if you can add serial monitor over network.

jonnycn, 2021-07-31:
Please do update~!

oli737, 2021-07-31:
Please update, isn't working anymore.

Froster69, 2021-07-31:
please update- ̗̀(๑ᵔ⌔ᵔ๑)

ktannady, 2021-07-31:
please update :D

panayotovip, 2021-07-31:
hey dudes look what i found :
https://github.com/AzagraMac/DuinoCoin_I2C
i do not tested that still

ricaun, 2021-07-31:
Yes, something is always changing on the main DuinoCoin project...

oli737, 2021-07-31:
> hey dudes look what i found :
> https://github.com/AzagraMac/DuinoCoin_I2C
> i do not tested that still

This one is not working you did not check the issue i opend on that.

ricaun, 2021-07-31:
The server is not sending jobs when the esp requests... 

I need to check the https://github.com/revoxhere/duino-coin/blob/master/AVR_Miner.py

panayotovip, 2021-07-31:
> 
> 
> > hey dudes look what i found :
> > https://github.com/AzagraMac/DuinoCoin_I2C
> > i do not tested that still
> 
> This one is not working you did not check the issue i opend on that.

did not have time when i found it.
thanks for testing.

panayotovip, 2021-08-01:
every thing is up and running ok
thankU-for-GJ ricaun

ericddm, 2021-08-02:
> every thing is up and running ok
> thankU-for-GJ ricaun

I got about 10 coins a day by 6 Arduino Nano,  is that ok?

ricaun, 2021-08-02:
> > every thing is up and running ok
> > thankU-for-GJ ricaun
> 
> I got about 10 coins a day by 6 Arduino Nano, is that ok?

I don't know, maybe the I2C is not so profit or the server is a little inconsistent.

ricaun, 2021-08-02:
The project now works on version 2.6.

closed 2021-08-02 by ricaun