ricaun/DuinoCoinI2C #31: error comiling version 2.6

panayotovip, 2021-08-01:
I delete old version, download new.
Load sketch in ArduinoIDE, then start compiling.
Did i do something wrong ?
Show me that output :
`Arduino: 1.8.15 (Windows 10), Board: "FireBeetle-ESP32, 80MHz, 115200"

C:\Users\IVAN\Downloads\DuinoCoinI2C-main\DuinoCoin_Esp_Async_Master\DuinoCoin_Clients.ino: In function 'bool clients_connect(byte)':

DuinoCoin_Clients:82:37: error: no matching function for call to 'WiFiClient::connect(String&, int&)'

   if (!clients[i].connect(host, port))

                                     ^

In file included from C:\Users\XXXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi\src/WiFi.h:37:0,

                 from C:\Users\XXXX\Downloads\DuinoCoinI2C-main\DuinoCoin_Esp_Async_Master\DuinoCoin_Esp_Async_Master.ino:27:

C:\Users\XXXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi\src/WiFiClient.h:51:9: note: candidate: virtual int WiFiClient::connect(IPAddress, uint16_t)

     int connect(IPAddress ip, uint16_t port);

         ^

C:\Users\XXXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi\src/WiFiClient.h:51:9: note:   no known conversion for argument 1 from 'String' to 'IPAddress'

C:\Users\XXXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi\src/WiFiClient.h:52:9: note: candidate: virtual int WiFiClient::connect(IPAddress, uint16_t, int32_t)

     int connect(IPAddress ip, uint16_t port, int32_t timeout);

         ^

C:\Users\XXXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi\src/WiFiClient.h:52:9: note:   candidate expects 3 arguments, 2 provided

C:\Users\XXXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi\src/WiFiClient.h:53:9: note: candidate: virtual int WiFiClient::connect(const char*, uint16_t)

     int connect(const char *host, uint16_t port);

         ^

C:\Users\XXXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi\src/WiFiClient.h:53:9: note:   no known conversion for argument 1 from 'String' to 'const char*'

C:\Users\XXXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi\src/WiFiClient.h:54:9: note: candidate: virtual int WiFiClient::connect(const char*, uint16_t, int32_t)

     int connect(const char *host, uint16_t port, int32_t timeout);

         ^

C:\Users\XXXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi\src/WiFiClient.h:54:9: note:   candidate expects 3 arguments, 2 provided

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\XXXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi

 Not used: C:\Users\XXXX\Downloads\arduino-1.8.15-windows\arduino-1.8.15\libraries\WiFi

exit status 1

no matching function for call to 'WiFiClient::connect(String&, int&)'`

sp4m4r, 2021-08-01:
Exactly the same issue here. ESP32 "NMCU-ESP32" with "ESP32 Dev Module" in Arduino IDE.

P.S. Thank you for the fantastic project, I'll spend 24 hours mining to your wallet once its running again! ;)

panayotovip, 2021-08-01:
> P.S. Thank you for the fantastic project, I'll spend 24 hours mining to your wallet once its running again! ;)
me too.

panayotovip, 2021-08-01:
> P.S.2 There's a typo in "D**o**inoCoin_AsyncWebServer.ino" ... maybe it's related?

need to add library 
https://github.com/me-no-dev/ESPAsyncWebServer

sp4m4r, 2021-08-01:
I added all new libraries, including the ESPAsyncWebserver (there is also a typo within the link "https://github.com/me-no-dev/ESPAsyncWebServer**D**"). I also updated the ESP32 Board to version 2.0.0-rc1, and tried to compile using different boards and settings - alway showing the same error as postet above.

ricaun, 2021-08-01:
> I added all new libraries, including the ESPAsyncWebserver (there is also a typo within the link "https://github.com/me-no-dev/ESPAsyncWebServer**D**"). I also updated the ESP32 Board to version [...]

I fix this misspell.

> I delete old version, download new.
> Load sketch in ArduinoIDE, then start compiling.
> Did i do something wrong ?
> Show me that output :
> `Arduino: 1.8.15 (Windows 10), Board: "FireBeetle-ESP32, 80MHz, [...]

I forgot to test with ESP32, I changed the `host` to String and this is the problem.

This should fix `host.c_str()` and return a `const char*`

Just update the line:

https://github.com/ricaun/DuinoCoinI2C/blob/cbde8045aac52a56db59f741b3b96582ac25a38c/DuinoCoin_Esp_Async_Master/DuinoCoin_Clients.ino#L82

closed 2021-08-01 by ricaun

sp4m4r, 2021-08-01:
That worked like a charm ... thank you so much!