ricaun/DuinoCoinI2C #32: Webserver question

Dark-Manu, 2021-08-05:
Hi @ricaun, 
I have installed & running the last version of your software and it works really well, (I only have some issues wit a NANO that hungs on "BAD jobs" until I restart it by hand, but my question is about the webserver feature.
After installing, I see that the port 80 TCP is opened in the board but I'm unable to access with the browser. 
Am I forgetting  any configuration to activate this feature? 
Also it could be that this is currently under development so, if it's the case, please forgot any of this message.
Thanks again for your work

ricaun, 2021-08-05:
Hello @Dark-Manu 

If the `BAD jobs` is consistent like the `BAD BAD BAD BAD` means you slave have some problem, try to reupload the last version.

Another thing could be the I2C clock too fast, I was testing to add oled display on the same I2C bus and need to set the speed clock to `100000`, the old library set the value too high and the slave loses some bytes and receives a wrong job...

You can try the oled tree. 

* https://github.com/ricaun/DuinoCoinI2C/tree/oled

About the Webserver: 

I'm still implement this feature, at the moment the home page is empty by default. The idea is to add some pages with some configuration about the slaves and DuinoCoin stuff.

You can access the page `/edit` to add some custom page, the user and password is 'admin' and 'admin'.

Use this `index.html` 
* https://gist.github.com/ricaun/366088002b08c98bb06873dd1d13f38f

Gonna add some like this:

[image]

Dark-Manu, 2021-08-06:
Wow!, thanks a lot for the info, it's very useful to me.
First, as you suggest, I'll check the speed of the I2C because I also was thinking on add an oled screen to easily check the rig status. Unfortunately, the link you attached, (https://github.com/ricaun/DuinoCoinI2C/tree/oled) takes me to this same project, main's page.
After that I'll mess with the webserver... it looks good.
I'm only a beginner in programming but projects like this help me to improving my skills, thanks again :)
I'll let you know if I have success.

ricaun, 2021-08-06:
Yes the link sends to a different tree.
https://github.com/ricaun/DuinoCoinI2C/tree/oled
The Esp folder have some new files tô support OLED display.

panayotovip, 2021-08-23:
did some one make index.html to look like more like Arduino IDE serial monitor log ?

sprokkie, 2021-11-22:
where to put the index file ?