ricaun/DuinoCoinI2C #33: update the picture with OLED and thanks to ricaun

ericddm, 2021-08-06:
very great job, thank you,ricaun.
[image]

panayotovip, 2021-08-07:
it look amazing !

pinokio240, 2021-08-07:
> very great job, thank you,ricaun.
> [image]

Здравствуйте. Вы сами делали эту плату? Я правельно понимаю что сдесь ESP8266? Кокая у Вас  операционная система и версия питна? Есть ли дополнительные программы кроме ArduinoIDE?

ericddm, 2021-08-07:
> > very great job, thank you,ricaun.
> > [image]
> 
> Здравствуйте. Вы сами делали эту плату? Я правельно понимаю что сдесь ESP8266? Кокая у Вас операционная система и версия питона? Есть ли дополнительные программы кроме ArduinoIDE?

Привет, мой друг. Я сам сделал эту доску. Он содержит один ESP8266 и 6 Mega328P. Мне нужно только установить ArduinoIDE, чтобы он заработал.
[ESP8266ArduinoNano6CoreDuinoMinerPCB.zip](https://github.com/ricaun/DuinoCoinI2C/files/6949361/ESP8266ArduinoNano6CoreDuinoMinerPCB.zip)

PCB files can be opened with Pads software,
_blz is a file that has been automatically routed and can be processed directly, 
and asc files can be imported with AltiumDesign.
[image]

ricaun, 2021-08-07:
> very great job, thank you,ricaun.
> [image]

Cool, @ericddm you should create a github project with pictures, schematic, files and so on.

closed 2021-08-07 by ricaun