ricaun/DuinoCoinI2C #34: Ошибка Arduino IDE

pinokio240, 2021-08-07:
Ошибка при компиляции:
  In file included from C:\Users\***\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Polyfills/static_array.hpp:11:0,
                 from C:\Users\***\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Numbers/FloatTraits.hpp:14,
                 from C:\Users\***\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Numbers/parseNumber.hpp:7,
                 from C:\Users\***\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Variant/VariantImpl.hpp:10,
                 from C:\Users\***\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson.hpp:32,
                 from C:\Users\***\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson.h:9,
                 from C:\Users\***\Desktop\DUCOcoint\DuinoCoinI2C-main\DuinoCoin_Esp_Async_Master\DuinoCoin_Pool.ino:15:
C:\Users\***\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In instantiation of 'typename ArduinoJson6183_91::enable_if<ArduinoJson6183_91::is_pointer<T>::value, T>::type ArduinoJson6183_91::pgm_read(const void*) [with T = const __FlashStringHelper*; typename ArduinoJson6183_91::enable_if<ArduinoJson6183_91::is_pointer<T>::value, T>::type = const __FlashStringHelper*]':
C:\Users\***\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Deserialization/DeserializationError.hpp:100:12:   required from here
C:\Users\***\Documents\Arduino\libraries\ArduinoJson-6.x\src/ArduinoJson/Polyfills/pgmspace_generic.hpp:14:45: error: 'const void*' is not a pointer-to-object type
   return reinterpret_cast<T>(pgm_read_ptr(p));
                                             ^
exit status 1
Ошибка компиляции для платы NodeMCU 0.9 (ESP-12 Module).

ricaun, 2021-08-07:
...

What version of the ArduinoJson library are you using...

I do not speak Russian, please write in English.

pinokio240, 2021-08-07:
> What version of the ArduinoJson library are you using...

I am using the latest update and latest version of Arduino IDE. Operating system Windows7x64. Python 3.8 is installed. I, like you, do not know English))

pinokio240, 2021-08-23:
> ...
> 
> What version of the ArduinoJson library are you using...
> 
> I do not speak Russian, please write in English.

The topic can be closed. Installed in Win8 emulator. Installed in it: Arduino1.8.0, Python 3.9, JAVA ver. 8u301, all libraries associated with the project. Can you post the code with LCD SSD1306 support?