ricaun/DuinoCoinI2C #37: Not connecting to server

oli737, 2021-08-10:
....
23:41:37.008 -> [0]Connecting to Duino-Coin server... with port 2812
23:41:37.055 -> [0]Connection failed.
23:41:37.055 -> [ ]FreeRam: 42000 I2C Connected[ 0 1 2 3 4 ]
23:41:42.052 -> [ ]FreeRam: 41976 I2C Connected[ 0 1 2 3 4 ]
23:41:47.051 -> [ ]FreeRam: 41976 I2C Connected[ 0 1 2 3 4 ]
23:41:52.051 -> [ ]FreeRam: 41976 I2C Connected[ 0 1 2 3 4 ]
23:41:57.053 -> [ ]FreeRam: 41976 I2C Connected[ 0 1 2 3 4 ]
23:42:02.062 -> [ ]FreeRam: 41976 I2C Connected[ 0 1 2 3 4 ]
23:42:07.011 -> [0]Connecting to Duino-Coin server... with port 2812
23:42:07.058 -> [0]Connection failed.
....

ricaun, 2021-08-11:
Yeeee! The Server is offline... I'm having the same problem connecting too.

Some users change the port and have success, I tried all the ports and did not have success. 😞

oli737, 2021-08-11:
With "original" Version 2.6 and port 2812 it works so the error must be something in this code.
I have some ESP running without i2c. They are working fine. Thats why i set the "BUG" label.

oli737, 2021-08-11:
Working again.
Ticket can be closed.

ricaun, 2021-08-12:
Yep the pool now is working!

closed 2021-08-12 by ricaun