ricaun/DuinoCoinI2C #40: i2c not worked

Gloomydemin, 2021-08-21:
...........
Connected to WiFi!
Local IP address: 192.168.0.11
[ ]Update beyond-pool-1 50.112.145.154 6000
[ ]FreeRam: 41776 I2C Connected[ ]
[ ]FreeRam: 41776 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41272 I2C Connected[ ]
[ ]FreeRam: 41488 I2C Connected[ ]
[ ]FreeRam: 41488 I2C Connected[ ]
[ ]FreeRam: 41488 I2C Connected[ ]
[ ]FreeRam: 41488 I2C Connected[ ]
[ ]FreeRam: 41488 I2C Connected[ ]
[ ]FreeRam: 41488 I2C Connected[ ]
[ ]FreeRam: 41488 I2C Connected[ ]
[ ]FreeRam: 41488 I2C Connected[ ]
[ ]FreeRam: 41488 I2C Connected[ ]
[ ]FreeRam: 41488 I2C Connected[ ]

how to fix?

dh1201, 2021-08-21:
The most probable cause for this is that you have an error in your wiring.
Carefully check the wiring - this is the most probable cause.  Also do not use crappy breadboard jumper type cables as these do not give a good enough connection in most cases.  It might also be necessary (depending on your set up) to include the pull up resistors.  (See here for a good write up on I2C: https://learn.sparkfun.com/tutorials/i2c/all

pinokio240, 2021-08-23:
> The most probable cause for this is that you have an error in your wiring.
> Carefully check the wiring - this is the most probable cause. Also do not use crappy breadboard jumper type cables as these [...]

I had the same problem until I updated the firmware to release from 08/23/21