ricaun/DuinoCoinI2C #41: Connection Failed

dh1201, 2021-08-21:
Morning,

I have a small 'rig' set up with 4 Arduino Nanos and an ESP8266 which has been running well, I have installed the latest Client and Server yesterday to try to solve my issue but have had no luck.

The ESP8266 connects correctly to the Wi Fi
The ESP8266 correctly identifies the 4 Nano 'slaves' `[ ] FreeRam: 41448 I2C Connected [ 0 1 2 3 ]`
Then cannot connect with:

```
Connected to WiFi!
Local IP address xxx.xxx.xxx.xxx
[ ]Update beyond-pool-1 50.112.145.154 6000
[0] Connecting to Duino-Coin server... with port 6000
[0] connection failed
```

Any ideas?

Thanks

panayotovip, 2021-08-23:
i do have that issue too.
i use esp32.
in DuinoCoin_Clients.ino is :
`String host = "149.91.88.18";
int port = 6002;`
but in DuinoCoin_Pool.ino
from http://51.15.127.80:4242/getPool
get other addres :
`name | "winner-pool-2"
ip | "193.164.7.180"
port | 6002
connections | 4688`

panayotovip, 2021-08-23:
try to edit.
no change.
same result:
12:32:09.733 -> [0]Connecting to Duino-Coin server... with port 6002
12:32:09.781 -> [0]Connection failed.
12:32:09.908 -> [ ]FreeRam: 242120 I2C Connected[ 0 1 2 3 ]
12:32:14.908 -> [ ]FreeRam: 242152 I2C Connected[ 0 1 2 3 ]
12:32:19.914 -> [ ]FreeRam: 242152 I2C Connected[ 0 1 2 3 ]
12:32:24.885 -> [ ]FreeRam: 242152 I2C Connected[ 0 1 2 3 ]
12:32:29.889 -> [ ]FreeRam: 242152 I2C Connected[ 0 1 2 3 ]
12:32:34.928 -> [ ]FreeRam: 242152 I2C Connected[ 0 1 2 3 ]
12:32:39.758 -> [0]Connecting to Duino-Coin server... with port 6002
12:32:39.803 -> [0]Connection failed.
12:32:39.885 -> [ ]FreeRam: 242120 I2C Connected[ 0 1 2 3 ]
12:32:44.908 -> [ ]FreeRam: 242152 I2C Connected[ 0 1 2 3 ]
12:32:49.901 -> [ ]FreeRam: 242152 I2C Connected[ 0 1 2 3 ]
12:32:54.921 -> [ ]FreeRam: 242152 I2C Connected[ 0 1 2 3 ]
12:32:59.909 -> [ ]FreeRam: 242152 I2C Connected[ 0 1 2 3 ]
12:33:04.885 -> [ ]FreeRam: 242152 I2C Connected[ 0 1 2 3 ]
12:33:09.753 -> [0]Connecting to Duino-Coin server... with port 6002
12:33:09.800 -> [0]Connection failed.

ricaun, 2021-08-23:
Yep I have this problem too. The server is too busy or something.

I add this line to kinda find another Pool and try to connect again in a different Pool.

https://github.com/ricaun/DuinoCoinI2C/blob/131ab186643fadcbdd80d6e474764f5c900b093c/DuinoCoin_Esp_Async_Master/DuinoCoin_Clients.ino#L86

ricaun, 2021-08-23:
This PR #42 update the typo mistake, the host variable never updates... 

Thanks @Vulpecula-nl.

dh1201, 2021-08-23:
Can confirm that once the new code is loaded it connects straight away.  Thanks for your help.

closed 2021-08-23 by dh1201