ricaun/DuinoCoinI2C #43: Question i2c other devices support

PipeDeveloper, 2021-08-24:
This is not actually a feature request but maybe could help to test some issues:

The scheme of this project is to use an ESP8266 or ESP32 as host master to connect arduinos to the server.
The issue is related that when you connect large ammounts of devices on the same i2c bus, they become in slow share frequency.
For example:
If you connect 3 devices, they do aprox 5000 shares per hour
When you connect 30 devices, they do aprox 2300 shares per hour

I would like to explore some alternatives about use a more powerfull host and do some testing.
Im wondering if i could use a raspberry pi 3-4 as a host, and use the i2c to comunicate with arduinos. The scheme should be the same, the only diference is how to run the code on a terminal.

Not related only a raspberry pi, but other ARM boards on linux based with i2c support, such orange pi, banana pi, etc.

pinokio240, 2021-08-24:
I will not write a new question.
   Why not program I2C on ESP for more Arduino board connectivity?
How will it affect esp and its functions?

ricaun, 2021-08-24:
> This is not actually a feature request but maybe could help to test some issues:
>
> The scheme of this project is to use an ESP8266 or ESP32 as host master to connect arduinos to the server.
> The issue [...]

I really don't know if a faster master solve the slow share issue with many slaves.

I believe is a i2c limitation. To many slaves waiting on the same bus and still need to send to the server.

I still thinks has some wrong with the code and master is waiting the slave to finish the job.

A good test should be add a esp8266 as a slave.

The job should be fast as crazy. 🤣

PipeDeveloper, 2021-08-24:
Let me test by mixing some diferent devices, but defenitely there is something weird when you connect many devices on the same bus. What i read is that the i2c bus can transmit from 100kb/s to an average 400kb/s. I think that is enough and pretty fast for this aplication.

My first suspicius responsible is the ESP8266, that doesnt have enough compute power to process incoming and transmit data