ricaun/DuinoCoinI2C #45: Please 🙏 update to version 2.7

oli737, 2021-08-30:
Version has updated to 2.7 and make AVR miners run again. Please update the code. Thank you.

ReoGustiono, 2021-08-30:
yes, I also hope that DuinoCoinI2C can be active again

panayotovip, 2021-08-30:
also +1 for update 

:( but in changelog ...
```
Release 2.7
Changelog:
    PC Miner:
        Full code rewrite
    AVR Miner:
        No longer open-source
        Arduino mining enabled again
    ESP8266 and ESP32 codes:
        Small improvements done over the past few days
```
I'm very worried about the part: 
```
AVR Miner:
   No longer open-source
   Arduino mining enabled again
```
[link to changelog](https://github.com/revoxhere/duino-coin/releases/tag/2.7)

ricaun, 2021-08-31:
Yep the AVR miner looks really crazy and change so many things...

I was able to make request to the server but not able to send job right, only received bad jobs.

😢

Gonna update the main code if someone wants to try to decode the AVR Miner to understand what's changed.

pinokio240, 2021-08-31:
> Yep the AVR miner looks really crazy and change so many things...
......
> Gonna update the main code if someone wants to try to decode the AVR Miner to understand what's changed.

And the news of today. They want to do closed source ((

ricaun, 2021-08-31:
> 
> 
> > Yep the AVR miner looks really crazy and change so many things...
> > ......
> > Gonna update the main code if someone wants to try to decode the AVR Miner to understand what's changed.
> 
> And the news of today. They want to do closed source ((

Yep when something involves money and anyone could create a miner and requesting jobs like an Arduino... 

Is hard to control...

panayotovip, 2021-08-31:
Your project ricaun is awesome.

ricaun, 2021-08-31:
Now is working!
I update to version 2.7.

closed 2021-08-31 by ricaun