ricaun/DuinoCoinI2C #48: ATMega328 change to difficulty @16

Dendrobats, 2021-09-03:
On version 2.7.

I have Arduino nano rig with esp8266 as master nad arduinos nano as slaves.
Made by from: https://github.com/ricaun/DuinoCoinI2C
Difficulty jumps from 6 to 16 and 20%++ bad shares.
Hasrate per device ~ 198.08 H/s

[image]

ricaun, 2021-09-03:
Thanks for sharing, probably some speed check was add on the main server, usually when your board is too fast the server adda difficult.

And the server is sending BAD jobs when the speed is too off.

Thays why I add a slow down speed to try to fix this problem.

ricaun, 2021-09-08:
Hello, I add the HASHRATE_SPEED to force the hash speed to be like195...

Is not the best but at the moment is the solution.

The server is off in this by the way....

closed 2021-09-08 by ricaun