ricaun/DuinoCoinI2C #49: Arduino Nano not recognized after power on with >6 devices.

fopinski, 2021-09-07:
**Describe the bug**
When there are more than 6 slave devices the system will not recognize these in 1 out of 10 power on cycles.
The device list is then just empty [ ].

**To Reproduce**
Steps to reproduce the behavior:
I use an ESP8266-01S as Master device and 10 Arduino Nano as Slaves.
Use more than 6 Arduinos.

**Expected behavior**
While I had only 6 Nano the rig worked immediately after power on.
When I added 4 additional Nano it works 1 out of 10 times after the power on.
The device list is just empty [ ] when the problem occurs.
It does not depend on the amount of maximum devices configured in the ESP.

**Arduino (please complete the following information):**
 - Arduino IDE Version 2 Beta
 - lates code from this github

**Additional context**
The wiring is correct. If it works it just works right after the power on, no issue then with the single Arduinos.

fopinski, 2021-09-12:
I managed to run 20 Arduino with the mentioned delay in the Arduino setup and a delay in the client_loop()  code of the esp. There has to be at least a 5ms delay in the end of the for loop.
ESP at 160MHz. i2c clock no faster than 100KHz.

fopinski, 2021-09-13:
Reward is 1 Duco per AVR.
Is it possible that the DuinoCoin library needs an update?

ricaun, 2021-09-13:
Hello!

This random makes sense, I implement something like this on the `DuinoCoin_setup` function.

But sometimes causes some bug and block all the slaves... That's why I remove the automatic I2C and set it to the fixed number using the eeprom on the tiny code.

> 
> 
> Reward is 1 Duco per AVR.
> Is it possible that the DuinoCoin library needs an update?

Probably the ESP is too slow to manage all the slaves at the same time and/or the code has some, some users report the rewards drop exponentially when having to many slaves on the same master.