ricaun/DuinoCoinI2C #50: can't connect to server

panayotovip, 2021-09-07:
**Describe the bug**
updated today after update can't connect to server :

**To Reproduce**
downloaded code, edit for my network and user, upload

**Expected behavior**
normal mining

**Screenshots**
```15:54:57.846 -> [0]2.7
15:56:37.814 -> [1]2.7
15:56:37.814 -> [ ]FreeRam: 177115 I2C [ 0 1 ]
15:56:39.879 -> [0]Connecting to Duino-Coin server... 149.91.88.18 2811
15:56:40.206 -> [1]Connecting to Duino-Coin server... 149.91.88.18 2811
15:56:40.438 -> [ ]FreeRam: 179223 I2C [ 0. 1. ]
15:58:21.376 -> [0]2.7
16:00:01.381 -> [1]2.7
16:00:01.381 -> [ ]FreeRam: 177047 I2C [ 0 1 ]
16:00:03.421 -> [0]Connecting to Duino-Coin server... 149.91.88.18 2811
16:00:03.467 -> [1]Connecting to Duino-Coin server... 149.91.88.18 2811
16:00:03.561 -> [ ]FreeRam: 179007 I2C [ 0. 1. ]
16:01:43.719 -> [0]2.7
16:02:43.115 -> ws[/ws][1] connect
16:02:43.255 -> ws[/ws][1] pong[0]: 
16:03:23.729 -> [1]2.7
16:03:23.729 -> [ ]FreeRam: 175859 I2C [ 0 1 ]
16:03:25.766 -> [0]Connecting to Duino-Coin server... 149.91.88.18 2811
16:03:28.778 -> [0]Connection failed.
16:03:28.872 -> [ ]Update pulse-pool-1 149.91.88.18 1224
16:03:28.872 -> [ ]FreeRam: 175379 I2C [ 0 1 ]
16:03:29.906 -> [ ]FreeRam: 175563 I2C [ 0 1 ]
16:03:30.893 -> [ ]FreeRam: 175559 I2C [ 0 1 ]
16:03:31.877 -> [ ]FreeRam: 175539 I2C [ 0 1 ]
16:03:32.911 -> [ ]FreeRam: 175543 I2C [ 0 1 ]
16:03:33.899 -> [ ]FreeRam: 175543 I2C [ 0 1 ]
16:03:34.887 -> [ ]FreeRam: 175543 I2C [ 0 1 ]
16:03:35.877 -> [ ]FreeRam: 175547 I2C [ 0 1 ]
16:03:36.910 -> [ ]FreeRam: 175539 I2C [ 0 1 ]
16:03:37.898 -> [ ]FreeRam: 175535 I2C [ 0 1 ]
16:03:38.886 -> [ ]FreeRam: 175555 I2C [ 0 1 ]
16:03:39.870 -> [ ]FreeRam: 175555 I2C [ 0 1 ]
16:03:40.891 -> [ ]FreeRam: 175555 I2C [ 0 1 ]
16:03:41.875 -> [ ]FreeRam: 175555 I2C [ 0 1 ]
16:03:42.891 -> [ ]FreeRam: 175555 I2C [ 0 1 ]
16:03:43.880 -> [ ]FreeRam: 175535 I2C [ 0 1 ]
16:03:44.915 -> [ ]FreeRam: 175527 I2C [ 0 1 ]
16:03:45.902 -> [ ]FreeRam: 175551 I2C [ 0 1 ]
16:03:46.889 -> [ ]FreeRam: 175539 I2C [ 0 1 ]
16:03:47.923 -> [ ]FreeRam: 175555 I2C [ 0 1 ]
16:03:48.897 -> [ ]FreeRam: 175555 I2C [ 0 1 ]
16:03:49.883 -> [ ]FreeRam: 175555 I2C [ 0 1 ]
16:03:50.917 -> [ ]FreeRam: 175555 I2C [ 0 1 ]
16:03:51.905 -> [ ]FreeRam: 175555 I2C [ 0 1 ]
16:03:52.881 -> [ ]FreeRam: 175555 I2C [ 0 1 ]
16:03:55.749 -> [0]Connecting to Duino-Coin server... 149.91.88.18 1224
16:03:55.843 -> [1]Connecting to Duino-Coin server... 149.91.88.18 1224
16:03:55.936 -> [ ]FreeRam: 177719 I2C [ 0. 1. ]
16:03:56.966 -> [ ]FreeRam: 178463 I2C [ 0. 1. ]
16:05:36.989 -> [0]2.7
16:07:16.961 -> [1]2.7
16:07:16.961 -> [ ]FreeRam: 175779 I2C [ 0 1 ]
16:07:19.029 -> [0]Connecting to Duino-Coin server... 149.91.88.18 1224
16:07:19.075 -> [1]Connecting to Duino-Coin server... 149.91.88.18 1224
16:07:19.167 -> [ ]FreeRam: 177899 I2C [ 0. 1. ]
16:08:59.261 -> [0]2.7
16:10:39.251 -> [1]2.7
16:10:39.251 -> [ ]FreeRam: 175751 I2C [ 0 1 ]
16:10:41.273 -> [0]Connecting to Duino-Coin server... 149.91.88.18 1224
16:10:41.367 -> [1]Connecting to Duino-Coin server... 149.91.88.18 1224
16:10:41.461 -> [ ]FreeRam: 177863 I2C [ 0. 1. ]
16:10:42.432 -> [ ]FreeRam: 177907 I2C [ 0. 1. ]
16:12:22.488 -> [0]2.7
16:14:02.472 -> [1]2.7
16:14:02.472 -> [ ]FreeRam: 175743 I2C [ 0 1 ]
16:14:04.488 -> [0]Connecting to Duino-Coin server... 149.91.88.18 1224
16:14:04.535 -> [1]Connecting to Duino-Coin server... 149.91.88.18 1224
16:14:04.628 -> [ ]FreeRam: 177935 I2C [ 0. 1. ]
16:15:45.038 -> [1]2.7
16:15:45.038 -> [ ]FreeRam: 175487 I2C [ 0. 1 ]
16:15:47.061 -> [0]Connecting to Duino-Coin server... 149.91.88.18 1224
16:15:47.249 -> [1]Connecting to Duino-Coin server... 149.91.88.18 1224
16:15:47.437 -> [ ]FreeRam: 178259 I2C [ 0. 1. ]
16:15:48.423 -> [ ]FreeRam: 178667 I2C [ 0. 1. ]
16:17:28.665 -> [0]2.7
16:19:08.654 -> [1]2.7
16:19:08.654 -> [ ]FreeRam: 175775 I2C [ 0 1 ]
16:19:10.663 -> [0]Connecting to Duino-Coin server... 149.91.88.18 1224
16:19:10.756 -> [1]Connecting to Duino-Coin server... 149.91.88.18 1224
16:19:10.849 -> [ ]FreeRam: 177883 I2C [ 0. 1. ]
16:19:11.855 -> [ ]FreeRam: 177931 I2C [ 0. 1. ]
16:20:52.414 -> [0]2.7
16:22:32.414 -> [1]2.7
16:22:32.414 -> [ ]FreeRam: 175743 I2C [ 0 1 ]
16:22:34.435 -> [0]Connecting to Duino-Coin server... 149.91.88.18 1224
16:22:34.575 -> [1]Connecting to Duino-Coin server... 149.91.88.18 1224
16:22:34.622 -> [ ]FreeRam: 177907 I2C [ 0. 1. ]
16:24:14.645 -> [0]2.7
16:25:54.657 -> [1]2.7
16:25:54.657 -> [ ]FreeRam: 175775 I2C [ 0 1 ]
16:25:56.679 -> [0]Connecting to Duino-Coin server... 149.91.88.18 1224
16:25:56.914 -> [1]Connecting to Duino-Coin server... 149.91.88.18 1224
16:25:57.008 -> [ ]FreeRam: 177907 I2C [ 0. 1. ]
16:27:37.727 -> [0]2.7
16:29:17.725 -> [1]2.7
16:29:17.725 -> [ ]FreeRam: 175759 I2C [ 0 1 ]
16:29:19.750 -> [0]Connecting to Duino-Coin server... 149.91.88.18 1224
16:29:19.796 -> [1]Connecting to Duino-Coin server... 149.91.88.18 1224
16:29:19.937 -> [ ]FreeRam: 177899 I2C [ 0. 1. ]
16:29:20.924 -> [ ]FreeRam: 177895 I2C [ 0. 1. ]
16:31:01.101 -> [0]2.7
16:32:41.091 -> [1]2.7
16:32:41.091 -> [ ]FreeRam: 175759 I2C [ 0 1 ]
16:32:43.115 -> [0]Connecting to Duino-Coin server... 149.91.88.18 1224
16:32:43.207 -> [1]Connecting to Duino-Coin server... 149.91.88.18 1224
16:32:43.300 -> [ ]FreeRam: 177911 I2C [ 0. 1. ]
16:34:23.728 -> [1]2.7
16:34:23.728 -> [ ]FreeRam: 175475 I2C [ 0. 1 ]
16:34:25.750 -> [0]Connecting to Duino-Coin server... 149.91.88.18 1224
16:34:25.797 -> [1]Connecting to Duino-Coin server... 149.91.88.18 1224
16:34:25.938 -> [ ]FreeRam: 176483 I2C [ 0. 1. ]
16:34:26.927 -> [ ]FreeRam: 178283 I2C [ 0. 1. ]```

**Arduino (please complete the following information):**
Arduino IDE 1.8.15
ESP32

fopinski, 2021-09-08:
Master server has an issue since yesterday. Pools had timeouts. Please check the status on discord.

ricaun, 2021-09-13:
Hello!

That's really strange... At the moment looks like is working fine.

The ip and port is defined by the server on the link: http://51.15.127.80:4242/getPool

Usually return something like this:

```json
{
  "name": "beyond-pool-3",
  "ip": "35.173.194.122",
  "port": 6001,
  "success": true
}
```

Try to reupload the esp code, you have some really strange ips and port...

```
149.91.88.18 1224
50.112.145.154 443
```

Usually, the port has something like 6000 to 6004... 

See ya!

panayotovip, 2021-09-13:
reinstalled Arduino IDE 1.8.15,
redownload and add Library Dependency for Esp8266/Esp32 - Master
edit "DuinoCoin_Esp_Async_Master.ino" for my network
upload and that is the result :(
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1100
ho 0 tail 12 room 4
load:0x40078000,len:12308
load:0x40080400,len:3076
entry 0x400805ec

Duino-Coin AVR I2C v2.7
Address: 1
Address: 2
Connecting to: DUCOFI
.....
Connected to WiFi!
Local IP address: 192.168.0.101
[ ]Update beyond-pool-2 35.173.194.122 6000
[0]Connecting to Duino-Coin server... 35.173.194.122 6000
[1]Connecting to Duino-Coin server... 35.173.194.122 6000
[ ]FreeRam: 179463 I2C [ 0. 1. ]
[ ]FreeRam: 179791 I2C [ 0. 1. ]
[0]2.7
[1]2.7
[ ]FreeRam: 177115 I2C [ 0 1 ]
[0]Connecting to Duino-Coin server... 35.173.194.122 6000
[1]Connecting to Duino-Coin server... 35.173.194.122 6000
[ ]FreeRam: 179199 I2C [ 0. 1. ]
[0]2.7
[1]2.7
[ ]FreeRam: 176835 I2C [ 0 1 ]
[0]Connecting to Duino-Coin server... 35.173.194.122 6000
[1]Connecting to Duino-Coin server... 35.173.194.122 6000
[0]2.7
[1]2.7
[ ]FreeRam: 176843 I2C [ 0 1 ]
[0]Connecting to Duino-Coin server... 35.173.194.122 6000
[1]Connecting to Duino-Coin server... 35.173.194.122 6000
[ ]FreeRam: 178995 I2C [ 0. 1. ]
[0]2.7
[1]2.7
[ ]FreeRam: 176831 I2C [ 0 1 ]
[0]Connecting to Duino-Coin server... 35.173.194.122 6000
[1]Connecting to Duino-Coin server... 35.173.194.122 6000
[ ]FreeRam: 179003 I2C [ 0. 1. ]
[ ]FreeRam: 178999 I2C [ 0. 1. ]

```
and that is where i stuck :(

panayotovip, 2021-09-13:
i load http://51.15.127.80:4242/getPool in browser
wait a little hit refresh, repeat this several times.
i get that in browser :
```
{"name":"pulse-pool-1","ip":"149.91.88.18","port":1224,"success":true}
{"name":"beyond-pool-3","ip":"35.173.194.122","port":6001,"success":true}
{"name":"beyond-pool-4","ip":"50.112.145.154","port":443,"success":true}
{"name":"beyond-pool-2","ip":"35.173.194.122","port":6000,"success":true}
{"name":"star-pool-1","ip":"51.158.182.90","port":6006,"success":true}
```

i try to manual connect to 
```
const char * name = "star-pool-1";
const char * ip = "51.158.182.90";
port = 6006; 
```

same result :(
swap ESP32 for another one, but ... same result :(

panayotovip, 2021-09-14:
relocate to other place for test with another isp,
replace Logic Level Converter for test
no success :(

will there be any updates to the latest version soon? 

```
12:08:36.672 -> configsip: 0, SPIWP:0xee
12:08:36.672 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:08:36.672 -> mode:DIO, clock div:1
12:08:36.672 -> load:0x3fff0030,len:1100
12:08:36.672 -> ho 0 tail 12 room 4
12:08:36.672 -> load:0x40078000,len:12308
12:08:36.672 -> load:0x40080400,len:3076
12:08:36.672 -> entry 0x400805ec
12:08:37.045 -> 
12:08:37.045 -> Duino-Coin AVR I2C v2.7
12:08:37.045 -> Address: 1
12:08:37.045 -> Address: 2
12:08:37.092 -> Connecting to: FreeNET
12:08:37.698 -> .......
12:08:40.730 -> Connected to WiFi!
12:08:40.730 -> Local IP address: 192.168.0.55
12:08:41.011 -> [ ]Update beyond-pool-3 35.173.194.122 6001
12:08:43.213 -> [0]Connecting to Duino-Coin server... 35.173.194.122 6001
12:08:43.494 -> [1]Connecting to Duino-Coin server... 35.173.194.122 6001
12:08:43.728 -> [ ]FreeRam: 179591 I2C [ 0. 1. ]
12:10:24.089 -> [0]2.7
12:12:04.076 -> [1]2.7
12:12:04.076 -> [ ]FreeRam: 177043 I2C [ 0 1 ]
12:12:06.126 -> [0]Connecting to Duino-Coin server... 35.173.194.122 6001
12:12:06.361 -> [1]Connecting to Duino-Coin server... 35.173.194.122 6001
12:12:06.594 -> [ ]FreeRam: 179131 I2C [ 0. 1. ]
12:13:47.077 -> [0]2.7
12:15:27.048 -> [1]2.7
12:15:27.048 -> [ ]FreeRam: 176931 I2C [ 0 1 ]
12:15:29.086 -> [0]Connecting to Duino-Coin server... 35.173.194.122 6001
12:15:29.318 -> [1]Connecting to Duino-Coin server... 35.173.194.122 6001
12:17:09.534 -> [0]2.7
12:18:49.530 -> [1]2.7
12:18:49.530 -> [ ]FreeRam: 176959 I2C [ 0 1 ]
12:18:51.534 -> [0]Connecting to Duino-Coin server... 35.173.194.122 6001
12:18:51.858 -> [1]Connecting to Duino-Coin server... 35.173.194.122 6001
12:18:52.090 -> [ ]FreeRam: 179147 I2C [ 0. 1. ]
```

panayotovip, 2021-09-14:
I am considering replacing my DevKit DOIT ESP-WROOM-32 as a master.
It is to many problems with it and to much offline time already.
Please give advice what board to look for? What boards do you use? 
Is it any chance to add support for some Linux SBC (orange pi, raspberry pi or other) as I2C master ?

ricaun, 2021-09-15:
> 
> 
> I am considering replacing my DevKit DOIT ESP-WROOM-32 as a master.
> It is to many problems with it and to much offline time already.
> Please give advice what board to look for? What boards do you use?

Hello, really strange... For some reason, the server is closing your connection after receiving the version.

Don't know what's happening in your case. 😭 

Are you able to mine use Arduino or ESP using the main [duino-coin](https://github.com/revoxhere/duino-coin) code?

>
>
> Is it any chance to add support for some Linux SBC (orange pi, raspberry pi or other) as I2C master ?

Unfortunately, I don't have the knowledge to create some master to these platforms. 😞

panayotovip, 2021-09-15:
> Are you able to mine use Arduino or ESP using the main [duino-coin](https://github.com/revoxhere/duino-coin) code?

yes, mining normal with main duino-coin code with ESP.
my Arduinos is pro mini, they too mine normal with usb ttl uart adapter.
gonna swap to some other board for master host.
now is time for searching options. what board do you use (brand, model, version)?

ricaun, 2021-09-18:
A basic ESP8266 should work...
Try to reupload the code, the server updates and break the code.
Now is working again!

fopinski, 2021-09-18:
ESP8266 was ok before at least for me but ESP32 still not working. That's what I get with current code:

21:04:28.859 -> [ ]FreeRam: 240396 I2C [ 0 1 2 ]
21:04:30.287 -> [ ]FreeRam: 240396 I2C [ 0 1 2 ]
21:04:31.684 -> [ ]FreeRam: 240400 I2C [ 0 1 2 ]
21:04:33.123 -> [ ]FreeRam: 240396 I2C [ 0 1 2 ]
21:04:34.526 -> [ ]FreeRam: 240396 I2C [ 0 1 2 ]
21:04:35.942 -> [ ]FreeRam: 240396 I2C [ 0 1 2 ]
21:04:37.378 -> [ ]FreeRam: 240396 I2C [ 0 1 2 ]
21:04:39.429 -> [0]Connecting to Duino-Coin server... 5.230.69.132 6002
21:04:39.429 -> [1]Connecting to Duino-Coin server... 5.230.69.132 6002
21:04:39.475 -> [2]Connecting to Duino-Coin server... 5.230.69.132 6002
21:06:19.526 -> [0]2.7
21:06:19.526 -> [0]MOTD Request: blah
21:07:59.638 -> [1]2.7
21:07:59.638 -> [1]Job Request: blah
21:09:39.683 -> [2]2.7
21:09:39.683 -> [2]Job Request: blah
21:09:40.245 -> [ ]FreeRam: 234576 I2C [ 0 1 2 ]
21:09:42.291 -> [0]Connecting to Duino-Coin server... 5.230.69.132 6002
21:09:42.337 -> [1]Connecting to Duino-Coin server... 5.230.69.132 6002
21:09:42.337 -> [2]Connecting to Duino-Coin server... 5.230.69.132 6002

fopinski, 2021-09-19:
It's hanging right here at the readStringUntil() function. It's realy slow. I read something on howto get around this with read() but always have reboots... need to tinker abit

```
void clients_waitRequestVersion(byte i)
{
  if (clients[i].available()) {
    String buffer = clients[i].readStringUntil(END_TOKEN);
    Serial.println("[" + String(i) + "]" + buffer);
    clients_state(i, DUINO_STATE_JOB_REQUEST);
    if (clientsMOTD) clients_state(i, DUINO_STATE_MOTD_REQUEST);
    clientsMOTD = false;
  }
}
```

JoJos1220, 2022-01-03:
Hey guys,

I got almost the same failure the last 3weeks! My ESP8266 Master was not able to mine any more!
Today I tryed to solve the Problem by deleting the Port specific request to the duinocoin server by:

**http://51.15.127.80/getPool**
instead of
**http://51.15.127.80:4242/getPool**

Mining is now good again!

Maybe for the future a good update comparing with the new Code realeases of the original duino-coin code project,
where no IP-Addresse is used anymore but instead a Domain is mentioned, to get Pool:
"https://server.duinocoin.com/getPool"

See ya!