ricaun/DuinoCoinI2C #52: Rig stop mining..

Daviedz, 2021-09-18:
my rig just stop mining last night, tried re uploading code n slave and master but still cannot mine.

09:12:48.335 -> Duino-Coin AVR I2C v2.7
09:12:48.335 -> Address: 1
09:12:48.335 -> Address: 2
09:12:48.335 -> Address: 3
09:12:48.335 -> Address: 4
09:12:48.335 -> Connecting to: AP_WiFi6
09:12:48.849 -> .......
09:12:52.583 -> Connected to WiFi!
09:12:52.583 -> Local IP address: 192.168.10.145
09:12:53.002 -> [ ]Update pulse-pool-4 149.91.88.18 2813
09:12:55.196 -> [0]Connecting to Duino-Coin server... 149.91.88.18 2813
09:12:55.474 -> [1]Connecting to Duino-Coin server... 149.91.88.18 2813
09:12:55.757 -> [2]Connecting to Duino-Coin server... 149.91.88.18 2813
09:12:56.037 -> [3]Connecting to Duino-Coin server... 149.91.88.18 2813
09:12:56.457 -> [0]2.7
09:12:56.551 -> [1]2.7
09:12:56.645 -> [2]2.7
09:12:56.645 -> [ ]FreeRam: 38568 I2C [ 0. 1. 2. 3. ]
09:12:56.690 -> [0]MOTD Request: Daviedz
09:12:56.690 -> [1]Job Request: Daviedz
09:12:56.690 -> [2]Job Request: Daviedz
09:12:57.344 -> [0]Happy mining on PulsePool (4)!
09:12:57.344 -> Pool worker limit: 50
09:12:57.344 -> [0]Job Request: Daviedz
09:12:57.675 -> [3]2.7
09:12:57.675 -> [ ]FreeRam: 38568 I2C [ 0. 1. 2. 3. ]
09:12:57.723 -> [3]Job Request: Daviedz
09:12:58.702 -> [ ]FreeRam: 38568 I2C [ 0. 1. 2. 3. ]
09:12:59.681 -> [ ]FreeRam: 38568 I2C [ 0. 1. 2. 3. ]
09:13:00.672 -> [ ]FreeRam: 38544 I2C [ 0. 1. 2. 3. ]
09:13:01.663 -> [ ]FreeRam: 38544 I2C [ 0. 1. 2. 3. ]
09:13:02.697 -> [ ]FreeRam: 38040 I2C [ 0. 1. 2. 3. ]
09:13:03.676 -> [ ]FreeRam: 38040 I2C [ 0. 1. 2. 3. ]
09:13:04.703 -> [ ]FreeRam: 38040 I2C [ 0. 1. 2. 3. ]
09:13:05.683 -> [ ]FreeRam: 38040 I2C [ 0. 1. 2. 3. ]
09:13:06.664 -> [ ]FreeRam: 38040 I2C [ 0. 1. 2. 3. ]
09:13:07.693 -> [ ]FreeRam: 38424 I2C [ 0. 1 2 3. ]
09:13:08.683 -> [ ]FreeRam: 38808 I2C [ 0 1 2 3 ]
09:13:09.676 -> [ ]FreeRam: 38808 I2C [ 0 1 2 3 ]
09:13:10.666 -> [ ]FreeRam: 38808 I2C [ 0 1 2 3 ]
09:13:11.702 -> [ ]FreeRam: 38808 I2C [ 0 1 2 3 ]
09:13:12.688 -> [ ]FreeRam: 38808 I2C [ 0 1 2 3 ]
09:13:13.675 -> [ ]FreeRam: 38808 I2C [ 0 1 2 3 ]
09:13:14.666 -> [ ]FreeRam: 38840 I2C [ 0 1 2 3 ]
09:13:15.703 -> [ ]FreeRam: 38832 I2C [ 0 1 2 3 ]
09:13:16.694 -> [ ]FreeRam: 38832 I2C [ 0 1 2 3 ]

Vulpecula-nl, 2021-09-18:
having the same problem

GridsETH2, 2021-09-18:
I'am the same problem. but for me i get port 6000

Connected to WiFi!
Local IP address: 192.168.1.57
[ ]Update svko-pool-1 5.230.69.132 6000
[0]Connecting to Duino-Coin server... 5.230.69.132 6000
[1]Connecting to Duino-Coin server... 5.230.69.132 6000
[2]Connecting to Duino-Coin server... 5.230.69.132 6000
[0]2.7
[1]2.7
[2]2.7
[ ]FreeRam: 38808 I2C [ 0. 1. 2. ]
[0]MOTD Request: GMiner1
[1]Job Request: GMiner1
[2]Job Request: GMiner1
[0]Welcome to svkopool (1) hosted by revox!
Pool worker limit: 50
[0]Job Request: GMiner1
[ ]FreeRam: 38808 I2C [ 0. 1. 2. ]
[ ]FreeRam: 38808 I2C [ 0. 1. 2. ]
[ ]FreeRam: 38808 I2C [ 0. 1. 2. ]
[ ]FreeRam: 38808 I2C [ 0. 1. 2. ]
[ ]FreeRam: 38304 I2C [ 0. 1. 2. ]
[ ]FreeRam: 38304 I2C [ 0. 1. 2. ]
[ ]FreeRam: 38304 I2C [ 0. 1. 2. ]
[ ]FreeRam: 38304 I2C [ 0. 1. 2. ]
[ ]FreeRam: 38304 I2C [ 0. 1. 2. ]
[ ]FreeRam: 38304 I2C [ 0. 1. 2. ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[ ]FreeRam: 38880 I2C [ 0 1 2 ]
[0]Connecting to Duino-Coin server... 5.230.69.132 6000

Carl1509, 2021-09-18:
Having the same issue with NodeMCU as host and Pro Minis as Slave since yesterday

ricaun, 2021-09-18:
Gonna fix, probably the server does not support the old not opensource code.

ricaun, 2021-09-18:
Should work!

Carl1509, 2021-09-18:
It works now thanks!

Daviedz, 2021-09-18:
> Should work!

Thanks

GridsETH2, 2021-09-19:
thanks it works

ricaun, 2021-09-21:
Gonna close. =D

closed 2021-09-21 by ricaun