ricaun/DuinoCoinI2C #53: TIMEOUT

rustamyou, 2021-09-18:
Duino-Coin AVR I2C v2.7.3
Address: 1
Address: 2
Connecting to: ROOter
.......
Connected to WiFi!
Local IP address: 192.168.1.135
[ ]Update svko-pool-1 5.230.69.132 6000
[0]Connecting to Duino-Coin server... 5.230.69.132 6000
[1]Connecting to Duino-Coin server... 5.230.69.132 6000
[0]2.7
[0]MOTD Request: rustamyou
[ ]FreeRam: 38688 I2C [ 0. 1. ]
[1]2.7
[1]Job Request: rustamyou
[0]Welcome to svkopool (1) hosted by revox!
Pool worker limit: 50
[0]Job Request: rustamyou
[1]Job 0a555cdb2cdf1f3fd97eba1d4b80a567da6fb932,c5fcd6d261ec7500c6a0b01f1432f0a28aacc5df,6
[1]Job Receive: 6
[ ]FreeRam: 38984 I2C [ 0. 1. ]
[0]Job ac827c882c69dd914910757762f4abe4edc87a85,e205b192d1d1ad69cc621c13ece071fc5dacaba1,6
[0]Job Receive: 6
[ ]FreeRam: 38536 I2C [ 0. 1. ]
[ ]FreeRam: 38536 I2C [ 0. 1. ]
[1] --------------- TIMEOUT -------------
[ ]FreeRam: 38400 I2C [ 0. 1 ]
[1]Connecting to Duino-Coin server... 5.230.69.132 6000
[0] --------------- TIMEOUT -------------
[1]2.7
[1]Job Request: rustamyou
[ ]FreeRam: 37536 I2C [ 0 1. ]
[1]Job c63f0df481eee1967d23231addc82962fa6e1467,9166b672de117fc0864fec32a462b236cccd2fa7,6
[1]Job Receive: 6
[ ]FreeRam: 38344 I2C [ 0 1. ]
[ ]FreeRam: 38344 I2C [ 0 1. ]
[ ]FreeRam: 38344 I2C [ 0 1. ]
[ ]FreeRam: 38344 I2C [ 0 1. ]
[ ]FreeRam: 38344 I2C [ 0 1. ]
[0]Connecting to Duino-Coin server... 5.230.69.132 6000
[1] --------------- TIMEOUT -------------
[ ]FreeRam: 37936 I2C [ 0. 1 ]
[0]2.7
[0]Job Request: rustamyou
[0]Job 4bd6462f35946b6f7875f203d16e4045129632c6,604db59d9beaa841b874e1ccefc4a47757248c32,6
[0]Job Receive: 6
[ ]FreeRam: 38152 I2C [ 0. 1 ]
[ ]FreeRam: 38152 I2C [ 0. 1 ]
[ ]FreeRam: 38152 I2C [ 0. 1 ]
[ ]FreeRam: 38152 I2C [ 0. 1 ]
[ ]FreeRam: 38152 I2C [ 0. 1 ]

Vulpecula-nl, 2021-09-18:
sEE #52

ricaun, 2021-09-18:
Looks like something related to the slave.
The slave never response back the job.

rustamyou, 2021-09-18:
> Looks like something related to the slave.
> The slave never response back the job.

Re: Vhhh
6 писем
UniqueID: 39 37 32 36 32 03 2A 18
Wire Address: 1
Job: 4ef7d7ca26cae2b215f28fa9e37bdaadd17cb8b3,67b410265e9ae7ddaf4d31cdc864424386f27f7a,6
Done 240,1341900,DUCOID3937323632032A18
Job: da1fdb967b4e484331dd9621fbf7831db13ac9c4,367f751f72ba951b95518409ebe3f4a13c33f585,6
Done 458,2563592,DUCOID3937323632032A18
Job: f8d1f65795f52d68331cc5135f049fc899728b26,1d510cb5865b88ec7065437af04c12894588cf57,6
Done 183,1027092,DUCOID3937323632032A18
Job: 461aba02d61eba3f411897fd961f4ce8e2db7474,6c5fea4e93776cfb7e422635f9d6ecc7cebd611b,6
Done 313,1753268,DUCOID3937323632032A18

nano executes but doesn't come to esp

ricaun, 2021-09-21:
Try to reupload the main code, I need to create some automatic updater.