ricaun/DuinoCoinI2C #57: LED on ATtiny85

Daviedz, 2021-09-23:
Does LED on Attiny85 off by default? I use Attiny as slave and only power LED is on but the other LED is not blinking when mining. How to turn it on?

fopinski, 2021-09-24:
https://github.com/ricaun/DuinoCoinI2C/blob/main/README.md

```
LED

This enable the led output and makes de led flash when the slave is working.

#define LED LED_BUILTIN
```
It depends on your board which Pin is used for the LED.

Daviedz, 2021-09-24:
> https://github.com/ricaun/DuinoCoinI2C/blob/main/README.md
> 
> ```
> LED
> 
> This enable the led output and makes de led flash when the slave is working.
> 
> #define LED LED_BUILTIN
> ```
> 
> It depends on your board which Pin is used for the LED.

Thanks, able to enable it..

closed 2021-09-24 by Daviedz