ricaun/DuinoCoinI2C #63: why can not work with arduino pro mini 168?

werdegars, 2021-10-30:
upload code successfully, no errors

board consist pro mini 328 (4 pcs), and pro mini 168 3.3v 8Mhz ( 6 pcs)
start up everything,
all 4 pro mini 328 working fine but none pro mini 168 is working
why?

malloc32, 2022-01-21:
I have the same problem, with 328p works fine, but with 168p (16MHz, 5V) not works, I have try with some of them.
Here you can see the log:
HOST:
23:03:04.720 -> [0]Job a99e50adede68b1d9a3f1b468794835ff6e6686d,a985323681c219aca549e8523b5fb5e1e1fe595f,4
23:03:04.720 -> [0]Job Receive: 4
23:03:05.523 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:06.521 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:07.512 -> [ ]FreeRam: 39248 I2C [ 0. ]
23:03:08.508 -> [ ]FreeRam: 39248 I2C [ 0. ]
23:03:09.545 -> [ ]FreeRam: 39248 I2C [ 0. ]
23:03:10.542 -> [ ]FreeRam: 39248 I2C [ 0. ]
23:03:11.538 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:12.531 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:13.529 -> [ ]FreeRam: 39248 I2C [ 0. ]
23:03:14.522 -> [ ]FreeRam: 39248 I2C [ 0. ]
23:03:15.515 -> [ ]FreeRam: 39248 I2C [ 0. ]
23:03:16.505 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:17.549 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:18.545 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:19.536 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:20.532 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:21.527 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:22.523 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:23.516 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:24.510 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:25.508 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:26.547 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:27.548 -> [ ]FreeRam: 39512 I2C [ 0. ]
23:03:28.540 -> [ ]FreeRam: 39512 I2C [ 0. ]
23:03:29.535 -> [ ]FreeRam: 39512 I2C [ 0. ]
23:03:30.531 -> [ ]FreeRam: 39512 I2C [ 0. ]
23:03:31.528 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:32.527 -> [ ]FreeRam: 39552 I2C [ 0. ]
23:03:33.520 -> [ ]FreeRam: 39248 I2C [ 0. ]
23:03:34.515 -> [ ]FreeRam: 39248 I2C [ 0. ]
23:03:34.752 -> [0] --------------- TIMEOUT ------------- 
23:03:35.507 -> [ ]FreeRam: 39248 I2C [ 0 ]
23:03:36.541 -> [ ]FreeRam: 39552 I2C [ 0 ]
23:03:37.536 -> [ ]FreeRam: 39552 I2C [ 0 ]

CLIENT (168P)

23:39:47.350 -> UniqueID: 30 35 35 34 31 18 10 02 
23:39:47.350 -> Wire Address: 1

this condition nevers appends   if (bufferReceive.available() > 0 && bufferReceive.indexOf('\n') != -1) { buffer>0 yes, but not the other.

Thank you

malloc32, 2022-01-22:
Can be a RAM problem? I have change code to use String instead StreamString and when string lenght is about 18 characters it stops to concatenate....

malloc32, 2022-01-22:
After trying to get moe and more memory the result was:
19:07:28.438 -> [0]0,194.51,AVR I2C v2.7.3,AVR-I2C-0
19:07:28.438 -> [ ]FreeRam: 38768 I2C [ 0. ]
19:07:28.484 -> [0]Job BAD,Incorrect result: Share #140 0:07:25
19:07:28.484 -> [0]Job Request: malloc32
19:07:28.532 -> [0]Job 165444811d1d658d7f338a5c835446631f3ffbd9,d29d9729288207265738c512e193a82c22f28a7a,4
So it does not work.
:(, I think that best solution is to use char[] instead String, but I have not the necessary knowledge.

Thank you.