ricaun/DuinoCoinI2C #64: I2C connection error

ThEnGI, 2021-11-14:
I cant work my rig, i think there is a i2c problem, i cheked the board and is connected right (the led powers on).
I'm an elettronic so i have very little C programming ability...
I have used the lastes library (downloaded yesterday)

The arduino works fine, 3 tried whit BLINK and one coming from another project, The esp32 also work fine becouse i used it for direct mining before. tried to flash evrithyng multiples time whit no solution.
One time it have worked (so so) but no work has started just a cicling connecting, no timeout  
i tried to chanmge the level converter whit a new one no solution, also tried whit less arduino's (even just one) no solution  
As picture: the circuit is composed ESP-32 -> Logic level converter -> 4*Arduino Nano. The power come from the usb of the ESP-32 then delivered to arduino. 

The serial monitor output:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1284
load:0x40078000,len:12836
load:0x40080400,len:3032
entry 0x400805e4

Duino-Coin AVR I2C v2.7.3
Connecting to: Riviera
.........
Connected to WiFi!
Local IP address: 192.168.0.109
[ ]Update svko-pool-1 5.230.69.132 6000
[ ]FreeRam: 226184 I2C [ ]
[ ]FreeRam: 225156 I2C [ ]
[ ]FreeRam: 226864 I2C [ ]
[ ]FreeRam: 227184 I2C [ ]
and so on 

```
a screen of the rig
[image]

ThEnGI, 2021-11-23:
Receck connection and LL connection, evrityng is correct.
Tried an old versione and che I2C works fine, the arduino is see by the ESP32 but i get unable to connect..... (is an old version so is normal :-) ) 
I think is a bug of this versione 2.7.3.....

Is possible to use another pin (+resistor + led) other then del led_builtin ? my esp32 dont have it !! 

[0]Connecting to Duino-Coin server... with port 6002
[0]Connection failed.
[ ]FreeRam: 230004 I2C Connected[ 0 1 2 4 ]
[ ]FreeRam: 230032 I2C Connected[ 0 1 2 4 ]
[ ]FreeRam: 230032 I2C Connected[ 0 1 2 4 ]

Saluti
ThEnGI

Sparwen, 2021-12-29:
Any news? I have the same problem, did you get this to work?

And also, have not the urlPool changed to https://server.duinocoin.com/getPool from http://51.15.127.80:4242/getPool?

Regards

ThEnGI, 2021-12-30:
unfortunately no, i'm still waiting for the update
I can't replay to your question :-(

Happy new year
ThEnGI

antoway, 2022-02-11:
Update? I've the same problem

Tuxglux, 2022-02-17:
Hi, I am on the same error since 2.7.3.
For the moment my rig works fine with A lab power supply. Not working like that with a normal USB-Power supply as it has done before 2.7.3. So maybe it is a problem in the power supply, like noise or instability.