ricaun/DuinoCoinI2C #68: ESP8266 stay in job request and then get job recieved: 1

omidtashakor, 2021-12-18:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Arduino (please complete the following information):**
 - Arduino IDE Version
 - Library's Version

**Additional context**
Add any other context about the problem here.

plofski, 2021-12-24:
same problem here, i cant seem to get a job from the server

PSS19871106, 2022-01-04:
new IP
String host = "176.32.35.94";

PSS19871106, 2022-01-04:
[image]