ricaun/DuinoCoinI2C #75: Connecting to Duino-Coin Server not possible

JoJos1220, 2022-01-14:
Hi Everyone!

I am not able to mine since today after a restart of the devices!
I always got frequently the Connection Error to the Duino-Coin Server.

_[0]Connecting to Duino-Coin server... 51.158.182.90 6000
[0]Connection failed.
[HTTP] GET... failed, error: connection failed_

May it be possible, that the IP-Address/Port is not any more in use?
Because pinging//Opening with Webbrowser and so on does not even deliver anything!

Kindly regards,

Jojo

Dev1Co, 2022-01-17:
In file DuinoCoin_Pool.ino  find the line (17) const char * urlPool = "http://51.15.127.80:4242/getPool"; and change to
//const char * urlPool = "http://51.15.127.80:4242/getPool";
const char * urlPool = "https://server.duinocoin.com/getPool";  or simply
const char * urlPool = "https://server.duinocoin.com/getPool";

Dev1Co, 2022-01-17:
I wonder if the project will be updated?

JoJos1220, 2022-01-18:
Thank you for your Replay,
after almost a half day of searching - not only changing the Pool address solves that problem at all!
I got the Workers back online after I changed the whole "UpdatePool" and the "httpgetstring" loop, because of the https -->> Secure connection!

Now it works again!

I will welcome the Update of the Project to 3.0! Maybe lets do it together!

Lanjelin, 2022-01-23:
> Thank you for your Replay, after almost a half day of searching - not only changing the Pool address solves that problem at all! I got the Workers back online after I changed the whole "UpdatePool" [...]

What a helpful comment! Congratulations to you!

For everyone else, replace the content of the `DuinoCoin_Pool.ino` with the following:

```
/*
  DuinoCoin_Pool.ino
  created 31 07 2021
  by Luiz H. Cassettari
*/

#if ESP8266
#include <ESP8266HTTPClient.h>
#include <WiFiClient.h>
#include <WiFiClientSecure.h>
#endif
#if ESP32
#include <HTTPClient.h>
#endif

#include <ArduinoJson.h>

//const char * urlPool = "http://51.15.127.80:4242/getPool";
const char * urlPool = "https://server.duinocoin.com/getPool";

void UpdateHostPort(String input)
{
  // {"ip":"server.duinocoin.com","port":2812,"name":"Main server"}
  DynamicJsonDocument doc(256);
  deserializeJson(doc, input);

  const char* name = doc["name"];
  const char* ip = doc["ip"];
  int port = doc["port"];

  Serial.println("[ ]Update " + String(name) + " " + String(ip) + " " + String(port));
  SetHostPort(String(ip), port);
}

void UpdatePool()
{
  String input = httpGetString(urlPool);
  if (input == "") return;
  UpdateHostPort(input);
}

String httpGetString(String URL)
{
  String payload = "";
  WiFiClientSecure client;
  client.setInsecure();
  HTTPClient http;
  if (http.begin(client, URL))
  {
    int httpCode = http.GET();
    if (httpCode == HTTP_CODE_OK)
    {
      payload = http.getString();
    }
    else
    {
      Serial.printf("[HTTPS] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
    }
    http.end();
  }
  return payload;
}
```

19leva89, 2022-01-30:
Good afternoon, I use your DUCO mining script, I have 6 Arduino Pro Mini and ESP32 boards, and I can’t get all 6 miners to work at the same time, there is a connection to the server, a request for work and all the work itself does not receive:( If in In the code, specify 3 miners, then everything works. All 6 boards receive an I2C address. Can you tell me what's wrong?