ricaun/DuinoCoinI2C #77: duino-coin  Release 3.0

MiloPS3, 2022-01-17:
Hi, are you planning to upgrade DuinoCoinI2C ? 

> _upgrading to this release is required. Servers will stop accepting version 2.X miners in a few weeks._

ricaun, 2022-01-17:
I need to take a look at the new Release. 😭

JoJos1220, 2022-01-18:
Lets do it!

JoJos1220, 2022-01-18:
> Hi, are you planning to upgrade DuinoCoinI2C ?
> 
> > _upgrading to this release is required. Servers will stop accepting version 2.X miners in a few weeks._

Maybe only changing the "Miner Version" in the I2C Software to 3.0 and thats it ;)

malloc32, 2022-02-01:
We can't mine using this project?

19:05:30.229 -> [ ]FreeRam: 39056 I2C [ 0. ]
19:05:30.839 -> [0]Job BAD,Outdated miner: Share #11 0:01:34

Thank you.

malloc32, 2022-02-01:
Thank you very much.

ricaun, 2022-02-01:
I don't know what changed... 😑

sabreman79, 2022-02-01:
Yeah, I guess they flipped the switch last night. I was mining happily, then BAM nothing. I'm a little dissapointed since I just finished my rig a couple of days ago :(

ricaun, 2022-02-01:
@revoxhere change some checks on the server, now should work.

The problem basically was related with the `rigIdentifier`.

closed 2022-02-01 by ricaun