ricaun/DuinoCoinI2C #82: Low Hashrate.

IgorHunterGames, 2022-02-05:
Hello, I installed the latest firmware on Arduino and connected via i2c to Wemos d1 mini. As a result, they give out a smaller hashrate than via USB. What could be the problem?
[image]

vaclavhorejsi, 2022-02-15:
Use this https://github.com/vaclavhorejsi/arduino-DuinoCoin instead ricaun´s DuinoCoin library

mt7725, 2022-02-27:
I've had the same issue. The problem was resolved using code from this link. https://github.com/JK-Rolling/DuinoCoinI2C . The slave code for all boards is the Attiny85 code. I've tried on Arduino Uno and Arduino Pro Mini boards and it works. And I've used vaclavhorejsi's DuinoCoin library files. Hope this helps. It worked for me.
[image]