ricaun/DuinoCoinI2C #84: DuinoCoin_Esp_Async_Master:30:10: fatal error: ArduinoOTA.h: No such file or directory

Vitorio-ui, 2022-02-15:
**Describe the bug**
I got this error while compile DuinoCoin_Esp_Async_Master

**To Reproduce**
Steps to reproduce the behavior:
1. I download it from https://github.com/ricaun/DuinoCoinI2C
2. Opened DuinoCoinI2C-main\DuinoCoin_Esp_Async_Master\DuinoCoin_Esp_Async_Master.ino
3. Go to "Sketch-Compile"
4. Error
DuinoCoin_Esp_Async_Master:30:10: fatal error: ArduinoOTA.h: No such file or directory
 #include <ArduinoOTA.h>
          ^~~~~~~~~~~~~~
compilation terminated.
exit status 1
ArduinoOTA.h: No such file or directory

**Expected behavior**
I expect to compile sketch

**Arduino (please complete the following information):**
 - Arduino IDE Version: 1.8.19 (windows store 1.8.57.0
 - Library's Version: DuinoCoinI2C -> DuinoCoin Version 3.0

vaclavhorejsi, 2022-02-15:
In Tools | Manage Libraries find ArduinoOTA and install it.

Vitorio-ui, 2022-02-15:
thanks

Next step
DoinoCoin_AsyncWebServer:20:10: fatal error: SPIFFSEditor.h: No such file or directory
 #include <SPIFFSEditor.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
SPIFFSEditor.h: No such file or directory

Vitorio-ui, 2022-02-15:
Thanks a lot.