ricaun/DuinoCoinI2C #85: does not connect to the server and does not show an error

limaarduino, 2022-02-19:
I'm using ESP32 as master and an Arduino Mega as slave.

after installing the current version of DuinoCoinI2C the serial monitor of the master shows the information below, looking like it has not connected to the server. There is no error return. And the message below appears in a loop.

MONITOR SERIAL MASTER - ESP32
11:04:11.621 -> Duino-Coin AVR I²C v3.0
11:04:11.668 -> Connecting to: Pegasus-2_2G
11:04:12.278 -> ......
11:04:14.760 -> Connected to WiFi!
11:04:14.760 -> Local IP address: 192.168.1.26
11:04:17.810 -> [ ]Update dream-pool-1 212.115.108.82 6000
11:04:17.998 -> [ ]FreeRam: 225580 I2C [ ]
11:04:19.027 -> [ ]FreeRam: 226056 I2C [ ]
11:04:20.012 -> [ ]FreeRam: 226056 I2C [ ]
11:04:20.998 -> [ ]FreeRam: 226056 I2C [ ]
11:04:22.032 -> [ ]FreeRam: 226056 I2C [ ]
11:04:23.014 -> [ ]FreeRam: 226056 I2C [ ]
11:04:24.002 -> [ ]FreeRam: 226056 I2C [ ]
11:04:25.036 -> [ ]FreeRam: 226056 I2C [ ]
11:04:26.020 -> [ ]FreeRam: 226056 I2C [ ]

MONITOR SERIAL DO SLAVE - ARDUINO MEGA
12:40:07.203 -> UniqueID: 34 34 31 38 35 10 0D 0C 
12:40:07.250 -> Wire Address: 1
12:46:59.963 -> UniqueID: 34 34 31 38 35 10 0D 0C 
12:46:59.963 -> Wire Address: 1

makukaka, 2022-03-04:
Can confirm / ESp32 Master - 2x Nano clones(AT168) 3x Nano clones(AT328p)
Please let me know if / when this will be fixed. :)

limaarduino, 2022-03-06:
> Can confirm / ESp32 Master - 2x Nano clones(AT168) 3x Nano clones(AT328p) Please let me know if / when this will be fixed. :)

I'm using this one and it worked. 

https://githubhot.com/repo/JK-Rolling/DuinoCoinI2C