ricaun/DuinoCoinI2C #90: Low profit with several miners

Sparwen, 2022-05-04:
Im getting nearly the same profit for 1 Arduino over I2C as 11 Arduinos over I2C?
Is that correct?

With 1 ATtiny1614 running 16Mhz im getting 250H/s hashrate and a calculated profit of 3-4 Ducos/day.
With 11 ATiny1614 running 16Mhz im getting total 2 730 H/s hashrate but only a calculated profit of 7 Ducos/day?

Shouln´t it be more like 20-30 Ducos/day, or at least 12 Ducos/day? (4 miners with 100% efficiency)

Regards

**To Reproduce**
Steps to reproduce the behavior:
1. Connect 1 miner, check calculated earnings.
2. Connect 11 miners, check calculated earnings.
3. Calculated earnings not multilied according to the kolka system

**Expected behavior**
20-30 Ducos/day, not 7 Ducos/day

**Arduino (please complete the following information):**
 - Arduino IDE 1.8.13 
 - DuinoCoinI2C 3.0 release

**Additional context**
Add any other context about the problem here.
[image]

NoNameNoCry, 2022-05-05:
[image]
[image]
hello.  same problem, additionally incorrect hashrate error

Sparwen, 2022-05-05:
The incorrect hashrate is just informing about that the hashrate is not between 240-270 H/s, wich is the "correct" hashrate for an Arduino Uno/Nano etc.

I dont think that it Effects the efficiency?

I tried 2 Arduino nano with the official AVR miner om the PC. Got about 20 Ducos/Day.

With the same arduinos running över I2C, i only getting about 7-8 Ducos/Day.

So it seems to be something wrong?

Sparwen, 2022-05-07:
After some further testing I could confirm that you don´t gain any profit with several slaves under the same ESP master.

I run 2 arduino slaves under 1 ESP master, and got around 7 Ducos/day.

Took the same arduinos but run them with seperate ESP masters ( 1master & 1slave x2 ) and got around 19 Ducos/day.

If this is the case, I can´t see the point of having several slaves under the same ESP master?

[image]
[image]

MiloPS3, 2022-05-09:
Hello, I have the same problem
1x master ESP8266 + 1 x I2C slave Attiny85 /16MHz/250H/s profit in 24 hours 4 Ducos (with kolka system 100%)
1x master ESP8266 + 10 x I2C slave Attiny85 /16MHz/2.58KH/s profit in 24 hours 2.7 Ducos (with limitations kolka system 8x81.54% + 1x78.28%)
from orientation table : Arduino Pro Mini, Uno, Nano etc.(Atmega 328p/pb/16u2) 	258 H/s 	0.2 W 	10-13 Duco/day , same hashrate but different profit
if there are a lot of slaves, they do not reach a high number of hashes and restart, kolka system will limit all devices

Sparwen, 2022-05-12:
So is this how it´s suppose to be?

hiddenotebook, 2022-10-26:
Someone finally found the solution to this issue?

Sparwen, 2022-11-03:
Nope I dont think so, its not a problem, i think its designed this way?

richie241, 2022-12-15:
Hey, are there any news for this topic?
Did you tested an other release? [https://github.com/JK-Rolling/DuinoCoinI2C](url)